stereolabs/zed-unity #4: ZED Unity integration assumes play space is always at origin

runevision, 2017-09-24:
Lots of VR experiences and games have ways of getting around in the world, whether with teleporting or virtual movement. The ZED integration doesn't seem to support this at all, as both the camera tracking logic and the garbage matte seems to assume that the play space is centered around the origin with no positional offset or rotation.

I managed to locally fix the tracking part by
- Moving the parent of the ZED_Greenscreen instance in under the [CameraRig] object.
- Modifying the ZEDPadManager script to record poses in world space rather than local play space by transforming the poses by the [CameraRig] transform as they are recorded.

I did not yet find a way to make the garbage matte work correctly.

pschuster, 2018-03-31:
It sucks that this repository is empty (no source code), makes it hard to do a Pull Request! However, I ran into the same problem, and fixed it used this code:

In *GarbageMatte.cs*:

Add this function which returns the transform of the SteamVR_PlayArea ([CameraRig]):

```
private Transform GetGarbageMatteParentTransform()
{
    SteamVR_PlayArea go = Object.FindObjectOfType<SteamVR_PlayArea>();
    return go.transform;
}
```

The GarbageMatte points (in the Editor) are sampled and saved in world space as you noted, too. I did not modify sampling but they are transformed to PlayArea local space when the planes are constructed.

Modify *GarbageMatte.cs* in _LoadData_ accordingly:
```
    public bool LoadData(GarbageMatteData garbageMatteData)
    {
        Transform playAreaTransform = GetGarbageMatteParentTransform();
       
        int nbMesh = garbageMatteData.numberMeshes;
        if (nbMesh < 0) return false;
            meshFilters[currentPlaneIndex].sharedMesh.MarkDynamic();
            Plane p = garbageMatteData.planes[i];
            for (int j = 0; j < p.numberVertices; j++)
            {                
                //<thisisnew>
                //Green Screen Editor samples Garbage Matte Points in World Space
                //Transform that world space points to local Play Area Space
                //Garbage Matte Planes are latter added as child to Play Area and will then
                //move with the Play Area when teleporting
                Vector3 worldSpacePoint = p.vertices[j];
                Vector3 localSpacePoint = playAreaTransform.InverseTransformPoint(worldSpacePoint);
                points.Add(localSpacePoint);
                //</thisisnew>
                GameObject sphere = GameObject.CreatePrimitive(PrimitiveType.Sphere);
                sphere.transform.localScale = new Vector3(0.2f, 0.2f, 0.2f);
		sphere.tag = "HelpObject";
                sphere.hideFlags = HideFlags.HideInHierarchy;
                outlineMaterial.SetFloat("_Outline", 0.00f);
                sphere.GetComponent<MeshRenderer>().material = outlineMaterial;

                //I left the sphere in world space (so I set position to the raw sampled world space points)
                sphere.transform.position = points[points.Count - 1];
                sphere.transform.position = p.vertices[j];   // <--- use p.vertices[I] instead of points
                sphere.layer = sphereLayer;
                spheresBorder.Add(sphere);
....
....
```

You should have the same, although you just sampled points differently. Planes are now part of PlayArea space and move along with it.

Now come the solution to your problem with Garbage Mattes not working. Garbage Mattes use a special shader that writes the planes to Stencil Buffer. Every pixel not in the Stencil Buffer is outside the Green Screen and is discarded.

However, ZED did modify the shader that it does not include the Transform of the plane into consideration. ZEDs implementation samples planes in world space, so that makes sense.

But we need to modify the shader so it works with our local implementation. 

Edit *Mask_Shader.shader*:
```
v2f vert(appdata v)
{
    v2f o;
    o.vertex = UnityObjectToClipPos(v.vertex);    //<--- add/uncomment this
    //o.vertex = mul(mul(UNITY_MATRIX_P, UNITY_MATRIX_V), v.vertex);  //<--- comment this
    o.uv = TRANSFORM_TEX(v.uv, _MainTex);
    return o;
}
```

Uncomment the first line (that is already in the shader but commented) and comment out that custom implementation added by ZED.

ZED just uses the camera view and projection matrix, but does leave out the local to world transformation matrix of the object.

Hope this helps.

@adujardin: Feel free to use the code as it's much more usable than only having Green Screen only compatible in one specific location.

And, please add the code to the repository so we can issue Pull Requests, as I have found quite a few bugs and issues I would like to contribute.

pschuster, 2018-04-26:
@adujardin Thank you very much for releasing the source to Github!

closed 2022-04-28 by github-actions[bot]