stereolabs/zed-unity #8: "ZED SDK is not installed or needs to be updated"

M4R5W0N6, 2017-12-11:
"The ZED plugin cannot be loaded" popup & "[ ZED Plugin ] : The SDK is not installed" error on play
meanwhile ZED_SDK_WinSetup_v2.2.1 has been launched and installed at C:\Program Files (x86)
the computer has been restarted as well

Unity 2017.3.0b11

obraun-sl, 2017-12-12:
Hi,
Did you check that the tools (ZED SDK/tools) are running OK ? ZED SDK 2.2.1 was provided with CUDA 8 and CUDA 9 compatibility. Did you choose the correct version?

ckorris, 2018-09-12:
@borderwing Two questions: 
1. Do the standard ZED SDK apps work correctly (Depth Viewer, etc. ) and does ZED Diagnostic say your system has the dependencies installed? 
2. Did you update an existing Unity project with a newer version? Sometimes Unity will restore or not overwrite old files if the upgrade is done improperly, resulting in the plugin checking for the wrong Unity version.

kevor, 2018-11-07:
Same problem here, 
Unity 2018.11f1
ZED SDK 2.7
GTX 1070
CUDA 10
nVidia Drivers 416.16

ZED Tools (like depth viewer) works correctly
[image]

obraun-sl, 2018-11-07:
@kevor,

Make sure you pull the last version of the unity plugin (v2.7) to be compatible with the ZED SDK (2.7).
You can find the unity package here (https://github.com/stereolabs/zed-unity/releases/tag/v2.7.0 ) or directly `git pull` to get the last commit

closed 2022-04-28 by github-actions[bot]