stereolabs/zed-unity #10: Depth transparency

renetto, 2018-01-25:
Hi,

I need to show the images from the camera in a determined range; in other words, I need to set alpha = 0 to pixels with depth > x.
I tried to set the DepthMaxRangeValue, but it affects only the depth texture, not the RGB one.

Is there a function in the SDK to do that?

Thank you,

R

WASCHMASCHINE, 2018-01-25:
No, there is no particular function to set alpha to 0 in the RGB image according to the max. depth range.

Still, it would be _trivial_ to loop through the depth image and adjust the same pixel coordinate according to your depth treshold, e.g. in OpenCV.

renetto, 2018-01-25:
Ok, thank you. I was trying to add a OnPreRender function() and call GetDepthValue() on each pixel, but, of course, the frame rate falls down.

Any idea about which script I should modify, or if I can modify the shader used for the "frame" (ZED Forward Lighting) to do that?

WASCHMASCHINE, 2018-01-26:
Oh my bad, I did not realize this was a Unity issue. You are right, CPU could would be bad for this.

I would suggest that you experiment with the shaders. I haven't looked into the current state of the Unity plugin and have no info about the shader code, but I suppose you could modify the pixel color (set your alpha) in a pixel shader where you can both sample the RGB texture and depth texture of the UV coordinate. I think it's entirely possible, you just need to tinker a bit.

obraun-sl, 2018-01-26:
Yes, using the CPU function won't be a good idea for performance.
The best is to modify zed_forward_lighting.shader to test outdepth value and modify outcolor.a with what you need. 
Note : this will work in forward rendering path. But same can be applied for deferred rendering

/*OB*/

renetto, 2018-02-07:
Yes, it works!

thank you @obraun-sl 
``` 
if(outDepth < 0.1) { //for some reasons, this means depth > 1m
    outColor.a = 0;
} else {
    outColor.a = 1;
}
```

SeiZon, 2018-02-16:
@renetto 
Any chance you would post your final result that works?
Could use the help myself. :)

Cheers

WASCHMASCHINE, 2018-02-16:
@renetto +1 for a picture :D

renetto, 2018-02-19:
Hi,

sorry for the delay! Here the final result ;)

@WASCHMASCHINE [image]

@SeiZon [https://gist.github.com/renetto/2026d9dff56ac46385220978c8f5d151](url)

closed 2022-04-28 by github-actions[bot]