stereolabs/zed-unity #14: Standalone plugin for C#

MiaoDX, 2018-05-07:
Dear devs,

Hi, thanks a lot for the various SDKs for the zed camera, I have used python/C++ SDK in my projects, and now, I am wondering if there are any plans for a standalone C# SDK instead of relaying the *heavy* unity3d as dependence?

Looking forward to your reply.

tomitrescak, 2018-06-06:
I am wondering the same. I tried to import the "sl_unitywrapper.dll" and "sl_mr_core.dll" into my project but no luck. Would you be able to provide wrappers for C# projects? For example to use it with EMGU.CV?

MiaoDX, 2018-06-27:
This seems to be somewhat solved (at least for me), please see [https://github.com/stereolabs/zed-unity/issues/15](https://github.com/stereolabs/zed-unity/issues/15) if you also need this, that is not the official version, but the codes pruned by @tomitrescak  is pretty good to use, I also pushed my local version to the github: [https://github.com/MiaoDX/ZED_EmguCV](https://github.com/MiaoDX/ZED_EmguCV), I removed the `async` codes to be more friendly to VS2010 (which is somewhat historical issue of my ongoing project). 

And thanks for @tomitrescak  !

closed 2018-06-27 by MiaoDX