stereolabs/zed-unity #15: ZEDMat to Image

tomitrescak, 2018-06-06:
Hi, that for the useful plugin!
I am trying to work with ZED SDK in C# and I made it as far as grabbing the Image into ZEDMat. I am trying now to process the image with EMGUCV, but I have **NO** Idea how to convert the ZEDMat to anything useful by C#.

I have created a GIST with the CODE: https://gist.github.com/tomitrescak/c31cee9c898ad7def32c4aaef8c1234d

The issue is marked with "???"

Would you be able to help the brother out?

obraun-sl, 2018-06-07:
Hi,

ZEDMat provides the public function MatPtr to access the data. Therefore I think you just need to copy the IntPtr from ZEDMat to Emgu.CV.Image using Marshal.copy()

Note that ZEDMat gives RGBA (4 channels) data. Your Emgu.CV.Image should be created in bgra if possible ( I'm not familiar with emgu.cv but since it's based on opencv, should be possible)

MiaoDX, 2018-06-17:
@tomitrescak , your example code seems promising, I am wondering:

1. Is it a plain C# project or an Unity one?
2. What is the `Reference` of the `using sl;`? I try to add `sl_mr_core64.dll`/`sl_unitywrapper.dll` from this repo or `sl_core64.dll`/`sl_zed64.dll` from the ZED sdk path, both failed: "A reference to ... could not be added. Please make sure that the file is accessible, and that it is a valid assembly or COM component".
3. Could you make it a standalone project?

Looking forward to your reply!

tomitrescak, 2018-06-17:
Hi, please see the code below thta controls ZED camera. Also, see screenshot with all the dependencies:

[image]

I have also gutted out the Unity package and created standalone SDK, also below:

```csharp
using Emgu.CV;
using Emgu.CV.Structure;
using sl;
using System;
using System.Collections.Generic;
using System.Threading;
using System.Threading.Tasks;

namespace Holoserver.Cameras
{
    public class Zed
    {
        ZEDCamera camera;
        InitParameters initParameters;
        object grabLock = new object();

        ERROR_CODE lastInitStatus = ERROR_CODE.ERROR_CODE_LAST;
        ERROR_CODE previousInitStatus;
        private bool running;
        private int stride;

        List<ZEDMat> mats;
        int matIndex;
        private RuntimeParameters runtimeParameters;

        // properties

        public event Action<Zed> FrameTick;

        public ZEDCamera Camera => this.camera;

        public RuntimeParameters RuntimeParameters { get => runtimeParameters; set => runtimeParameters = value; }

        public Image<Bgra, byte> Left => this.FetchImage(VIEW.LEFT);
        public Image<Bgra, byte> Right => this.FetchImage(VIEW.RIGHT);
        public Image<Bgra, byte> Depth => this.FetchImage(VIEW.DEPTH);
        public Image<Bgra, byte> Confidence => this.FetchImage(VIEW.CONFIDENCE);
        public Image<Bgra, byte> LeftGrey => this.FetchImage(VIEW.LEFT_GREY);
        public Image<Bgra, byte> RightGrey => this.FetchImage(VIEW.RIGHT_GREY);
        public Image<Bgra, byte> SideBySide => this.FetchImage(VIEW.SIDE_BY_SIDE);

        // constructor

        public Zed(string svoPath) : this(null, svoPath) { }

        public Zed(InitParameters parameters = null, string svoPath = null)
        {
            camera = ZEDCamera.GetInstance();

            mats = new List<ZEDMat>();

            if (parameters == null)
            {
                parameters = new InitParameters();
                parameters.resolution = RESOLUTION.HD720;
                parameters.depthMode = DEPTH_MODE.MEDIUM;
                parameters.depthStabilization = true;
                parameters.enableRightSideMeasure = true; // isStereoRig;
                parameters.depthMinimumDistance = 0.2f;
            }

            if (svoPath != null)
            {
[... 3335 lines cut ...]
        /// </summary>
        /// <param name="value"></param>
        /// <param name="mem"></param>
        /// <returns></returns>
        public sl.ERROR_CODE SetTo(ref char3 value, sl.ZEDMat.MEM mem)
        {
            return (sl.ERROR_CODE)(dllz_mat_set_to_uchar3(_matInternalPtr, ref value, (int)(mem)));
        }

        /// <summary>
        /// ills the Mat with the given value.
        /// </summary>
        /// <param name="value"></param>
        /// <param name="mem"></param>
        /// <returns></returns>
        public sl.ERROR_CODE SetTo( ref char4 value, sl.ZEDMat.MEM mem)
        {
            return (sl.ERROR_CODE)(dllz_mat_set_to_uchar4(_matInternalPtr, ref value, (int)(mem)));
        }
        /***************************************************************************************/

    }
}

```

MiaoDX, 2018-06-18:
WoW, that is a big effort! Unfortunately, still, I did not run it successfully. 

* I separate the SDK content into four files (`ZEDCamera.cs`,`ZEDCameraSettingsManager.cs`,`ZEDCommon.cs`,`ZEDMat.cs`) as in the official SDK and put them into `NativeInterface`. And put the `Zed.cs` into `Cameras` folder.
* After some tweaks the code build OK
* When running, first it was 
```
An unhandled exception of type 'System.DllNotFoundException' occurred in ZedTester.exe

Additional information: Unable to load DLL“sl_unitywrapper” The specified module could not be found. (Exception from HRESULT: 0x8007007E)
```

* Then I put `sl_mr_core64.dll` and `sl_unitywrapper.dll` into the Release folder, but the error now becomes

```
An unhandled exception of type 'System.BadImageFormatException' occurred in ZedTester.exe

Additional information: An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B)
```

* What is more, I have to comment out the `public RuntimeParameters RuntimeParameters { get => runtimeParameters; set => runtimeParameters = value; }` in order to build

I am sure there are some little errors here and there, so I am wondering, could it be possible that you email me (or other ways, maybe set up one repo in your github) one complete/standalone/clean project? Maybe filter out the private codes like the server project above?

Sorry for the trouble, my email addr: miaodx@tju.edu.cn

Thanks ahead!

MiaoDX, 2018-06-18:
@tomitrescak  It seems that I made it! Use the latest SDK (v2.4.1) with cuda9.1 and the dlls from this repo's v2.4.0 tag. The acquisition seems ok but still, `public RuntimeParameters RuntimeParameters { get => runtimeParameters; set => runtimeParameters = value; }` needs to be commented out in `class Zed`.

tomitrescak, 2018-06-18:
@MiaoDX make sure you are building with x64 target

And then observe the magic! ;)

tomitrescak, 2018-06-18:
Example:

```csharp
// Form1.cs
using sl;
using System;
using System.Windows.Forms;
using Holoserver.Cameras;
using System.Diagnostics;

namespace ZedTester
{

    public partial class Form1 : Form
    {

        VIEW view;

        public Form1()
        {
            InitializeComponent();

            // find version
            var versionZED = "[SDK]: " + sl.ZEDCamera.GetSDKVersion().ToString() + " [Plugin]: " + sl.ZEDCamera.PluginVersion.ToString();
            this.versionInfo.Text = versionZED;

            this.Start();
        }

        void Start()
        {
            // start with stored SVO
            var zed = new Zed(@"c:\Projects\files\HD720_SN17600_11-04-52.svo");

            // start with live feed
            // var zed = new Zed();

            // You need to assign the listener that will execute in each camera loop
            // in this case I named it AcquireImages and implemented it below
            zed.FrameTick += this.AcquireImages;
            zed.RuntimeParameters = new RuntimeParameters
            {
                sensingMode = SENSING_MODE.FILL,
                enableDepth = true
            };
            zed.Start();
        }

        private void AcquireImages(Zed zed)
        {
            try
            {
                imageBox1.Invoke((MethodInvoker)delegate
                {
            // you can use method FetchImage or any of the helper methods
            // e.g. zed.Left, zed.Right, zed.Depth ...
            var image = zed.FetchImage(view);
                    imageBox1.Image = image;
                });
            }
            catch { }

        }

        private void button1_Click(object sender, EventArgs e)
        {
            MessageBox.Show("Is: " + ZEDCamera.CheckPlugin());
        }

        private void sourceSelection_SelectedIndexChanged(object sender, EventArgs e)
[... 130 lines cut ...]
            this.AutoScaleDimensions = new System.Drawing.SizeF(6F, 13F);
            this.AutoScaleMode = System.Windows.Forms.AutoScaleMode.Font;
            this.ClientSize = new System.Drawing.Size(998, 664);
            this.Controls.Add(this.imageBox1);
            this.Controls.Add(this.label1);
            this.Controls.Add(this.sourceSelection);
            this.Controls.Add(this.errorInfo);
            this.Controls.Add(this.versionInfo);
            this.Name = "Form1";
            this.Text = "Form1";
            ((System.ComponentModel.ISupportInitialize)(this.imageBox1)).EndInit();
            this.ResumeLayout(false);
            this.PerformLayout();

        }

        #endregion
        private System.Windows.Forms.Label versionInfo;
        private System.Windows.Forms.Label errorInfo;
        private System.Windows.Forms.ComboBox sourceSelection;
        private System.Windows.Forms.Label label1;
        private Emgu.CV.UI.ImageBox imageBox1;
    }
}
```

MiaoDX, 2018-06-22:
@tomitrescak  Sorry to bother you again -.-

But could you email me one of the SDK exe you used, maybe it should be `ZED_SDK_WinSetup_v2.4.0.exe` (which is also the unity plugin), instead of v2.4.1 which is provided by the official website? At first, I mixed them but it still can work, but when I change to another computer it complains about the mismatch and terminated at the camera init phase. 

My email addr: miaodx@tju.edu.cn 

Looking forward to your reply!

And @nesnes @adujardin , I also report this issue (not all legacy SDKs are kept in the website and some other minor issues) to the 
 support@stereolabs.com , hope you devs can make things easier :)

closed 2022-04-28 by github-actions[bot]