stereolabs/zed-unity #17: Playing Example Crashes Unity

blairlc, 2018-07-12:
I have had success making Unity apps that use the ZED and Rift in the past. Now, trying to play any app that uses the ZED in Unity causes Unity to crash. 

Steps to reproduce:
1. Create a new 3D project in Unity 2017.3.1f1 or 2018.1.5f1.
2. Import ZED Plugin for Unity 2.4.1. Allow Unity to make updates.
3. Enable VR.
4. Delete the main camera and replace it with either of the ZED prefabs.
5. Press play.

The loading screen appears briefly and then the editor closes. I'm not sure which change is causing this problem. I have updated my Unity versions recently and I was also using the 2.4 ZED plugin before. It could also be caused by a Windows or Oculus update. I do not see any explanation in the Unity log.

blairlc, 2018-07-13:
I have done some more testing. The ZED example apps run fine, so the problem is not connecting to the camera. I tried using the Unity examples on another computer with the same version of Windows and it ran fine there. That computer still had ZED version 2.4.0, so I uninstalled 2.4.1 on the computer having the problem and reinstalled 2.4. Unity still crashes. I am going to see if upgrading to 2.5 solves the problem.

blairlc, 2018-07-13:
After upgrading to 2.5, the examples now work with an original ZED, but not with the mini. I suspect this was the problem the whole time as the working computer was using an original ZED.

derzorngottes, 2018-10-03:
I am having the same issue with using the ZED SDK and plugin version 2.4.1. I have reinstalled my graphics drivers and CUDA 9.1, and made sure I downloaded the ZED SDK targeting 9.1. I have another component which only works with 9.1 so updating the ZED SDK is not an option.

There is no debugging info in the Unity Editor logs and I cannot diagnose a cause for this crash.

derzorngottes, 2018-10-10:
I diagnosed this issue: I did a clean install of everything with the newest SDK. Then when I ran the diagnostic tool it indicated an issue with the USB driver for the port that the ZED was plugged into. This issue was not detected by the diagnostic tool when 2.4.1 was installed. I reinstalled the driver for the USB port and then reinstalled the 2.4.1 SDK and it worked.

TheOnlyUsernameThatWasStillAvailable, 2018-10-23:
I recently started working with ZED mini camera and I have a very similar issue: when I create even the simplest Unity project with ZED plugin, Unity freezes a few seconds after pressing the play button.

I'm using ZED Mini with latest firmware. I installed fresh NVidia drivers and CUDA 9.2. I'm using SDK version 2.6 and Unity plugin version 2.6. I tried two Unity versions: the latest (2018.2.13f1 64-bit) and the recommended 2017.4.13f1 (64-bit). All this runs on a Windows 7 64-bit with GTX 770 graphics card, I7-6700K processor and 32 GB RAM.

All the examples in the ZED SDK run fine and without problems. The ZED diagnostics program doesn't show any errors, although my graphics card is a little slow. Even the ZED World AR application runs fine using the Oculus Rift. 

But when I create a new Unity project, remove the original camera, add the ZED_Rig_Mono camera and then press Play, I can see the video from camera for a few seconds and then Unity freezes. The blue LED on the ZED mini stays lit until I unplug it.

Interestingly, if I set the ZED_Rig_Mono's Depth mode to None, then I can see the video and everything works fine. Any tips on how to fix this are greatly appreciated!

TheOnlyUsernameThatWasStillAvailable, 2018-10-24:
Thank you very much obraun-sl for the quick reply! I installed the driver version you recommended and now everything works! I'm really looking forward to start playing with the AR!