stereolabs/zed-unity #18: Missing distortion coefficients

shaharlevy29, 2018-07-15:
I'm trying to get the matrix parameters (ie calibration matrix and distortion coefiicients).
The script:

 sl.CalibrationParameters clb = zedCamera.GetCalibrationParameters();
float[] zedCamMat = { clb.leftCam.fx, clb.leftCam.cx, clb.leftCam.fy, clb.leftCam.cy};
double[] zedDistMat = clb.leftCam.disto;

returns values to zedCamMat (697.5081, 660.2579, 697.5081, 92.94684)
but the the zedDistMat is all zeroes.

How come? 

thanks,
Shahar

obraun-sl, 2018-07-16:
Hi,

Since we provide the rectified images, the distortion is indeed null since it's corrected. 
If you still want the raw distortion parameters (before correction), just call `zedCamera.GetCalibrationParameters(true);`

shaharlevy29, 2018-07-16:
Hi,
Thanks, I now understand that the disortion parameter should be zero. My application works fine. 
Thanks you!

BTW, the _zedCamera.GetCalibrationParameters(true);_ command returns zero on all parameters (both distortion and calibration matrix).

obraun-sl, 2018-07-16:
Ok I'll check for the raw parameters

closed 2022-04-28 by github-actions[bot]