stereolabs/zed-unity #19: No video displayed on fram

mohamad-issa, 2018-07-19:
Hello everyone,

I run the examples on Unity 2018.1.3f1, using ZED SDK 2.5 and Zed-Unity 2.5, the tracking works but I don't see any images.
In the ZED Explorer everything works fine.
Thanks for your help!

ckorris, 2018-09-12:
@mohamad-issa Hope this isn't too late. Does this happen in the example scenes? 

If so, I suspect another camera is covering up the final output to the ZED. I'd make sure you don't have another camera in the scene (like the default Main Camera that's added to every new scene) and that you have the stock ZED_Rig_Mono or ZED_Rig_Stereo prefab in the scene. If you need to have another camera in the scene, make sure its depth value is set to be lower than the camera(s) on your chosen ZED rig.

tonytsai036, 2019-03-25:
Hi,

I have the same problem. I run the simpleMR example with ZED(not mini), using ZED SDK 2.7.1 and Zed-Unity 2.7.0_b2, also the tracking works fine but no image render. And I only have ZED_Rig_Mono prefab in the scene.
[image]
[image]
Is their any step that I didn't notice?

Thanks for the help!

zvika-waissbaum, 2021-03-31:
Hi,

I have the same problem.
No image rendering.
I don't have any other camera.

Unity 2019.3.15f1
Latest ZED SDK with CUDA 11.2

Is there a solution?

closed 2022-04-28 by github-actions[bot]