stereolabs/zed-unity #25: Marshal.Copy(ZEDMat.MatPtr, byte[] t, 0, length) crashes?

axing0805091, 2018-09-13:
`using System.Collections;
using System.Collections.Generic;
using System.Runtime.InteropServices;
using UnityEngine;

public class textureManaging : MonoBehaviour {
    public GameObject ZED_Rig_Stereo;
    sl.ZEDMat slMat = new sl.ZEDMat(1280, 720, sl.ZEDMat.MAT_TYPE.MAT_8U_C4, sl.ZEDMat.MEM.MEM_CPU);
    ZEDManager zedmanager;
    byte[] t = new byte[1280 * 720 * 4];
    void Awake()
    {
        zedmanager = ZED_Rig_Stereo.GetComponent<ZEDManager>();
    }
    // Update is called once per frame
    void Update()
    {

        zedmanager.zedCamera.RetrieveImage(slMat, sl.VIEW.LEFT, sl.ZEDMat.MEM.MEM_CPU, slMat.GetResolution());
        Debug.Log(slMat.GetChannels());
        Debug.Log(slMat.GetInfos());
        Marshal.Copy(slMat._matInternalPtr, t, 0, 1280 * 720* 4);
    }
}`

I put a plane in the scene attached with the above script. And I put a ZED_Rig_Stereo prefab in the scene and attach it to the "public GameObject ZED_Rig_Stereo".  When I comment the last line code "Marshal.Copy()", it works well. When I uncomment it, Unity3D will shut down automatically. How can I transform ZEDMat into byte[]? I'm a beginner to c#, thanks for answering.

axing0805091, 2018-09-13:
By the way, I changed ZEDMat._matInternalPtr to public.  "Marshal.Copy(slMat.MatPtr, t, 0, 1280*720*4)" also has the same error.

axing0805091, 2018-09-13:
I want to transform ZEDMat into OpenCvSharp Mat to detect aruco markers, can anyone help me ?

dogod621, 2019-01-17:
I think may be _matInternalPtr is pont to ZEDMat object,
It is not pont to data,
it only contains width, height, type, dataPtr... etc informations.

Maybe you should use ZEDMate's GetPtr:

public System.IntPtr GetPtr(MEM mem = MEM.MEM_CPU)
{
            return dllz_mat_get_ptr(_matInternalPtr, (int)mem);
}

Then copy from this pointer (point to data)

closed 2022-04-28 by github-actions[bot]