stereolabs/zed-unity #28: SteamVR plugin + Zed + Unity 2018.2.13f1

philfree, 2018-10-24:
As soon as I install the SteamVR Plugin I get those errors in the Unity ZED / SDK
```
Assets/ZED/Examples/Plane Detection/Scripts/BunnyPlacement.cs(44,5): error CS0246: The type or namespace name `SteamVR_Controller' could not be found. Are you missing an assembly reference?
Assets/ZED/SDK/Helpers/Scripts/Interactions/ZEDSteamVRControllerManager.cs(128,10): error CS0246: The type or namespace name `SteamVR_ControllerManager' could not be found. Are you missing an assembly reference?
Assets/ZED/SDK/Helpers/Scripts/Interactions/ZEDSteamVRControllerManager.cs(134,14): error CS0246: The type or namespace name `SteamVR_TrackedController' could not be found. Are you missing an assembly reference?
Assets/ZED/SDK/Helpers/Scripts/Interactions/ZEDSteamVRControllerManager.cs(146,14): error CS0246: The type or namespace name `SteamVR_Controller' could not be found. Are you missing an assembly reference?
Assets/ZED/SDK/Helpers/Scripts/Interactions/ZEDSteamVRControllerManager.cs(117,10): error CS0246: The type or namespace name `SteamVR_Controller' could not be found. Are you missing an assembly reference?
```

ckorris, 2018-10-24:
Hi Philfree, I'll take a look tomorrow morning to see if the latest SteamVR plugin refactored something. But in the meantime, ZEDSteamVRControllerManager is deprecated so you can delete the whole class, and BunnyPlacement is required only for the VR plane detection sample so you can delete the whole VR Plane Detection example folder if you don't need it.

ckorris, 2018-10-24:
It looks like they indeed changed the SteamVR plugin last month to use a new Input system. We'll have to update that on our end to make it work. 

However, the old version of the SteamVR plugin will still work, downloadable here: https://github.com/ValveSoftware/steamvr_unity_plugin/releases/tag/1.2.3

philfree, 2018-10-24:
That worked, thanks

obraun-sl, 2018-10-26:
Merged into master.

closed 2018-10-26 by obraun-sl

ckorris, 2018-10-30:
@philfree FYI we changed things a bit so that it'll work in both the old and the new version of SteamVR. However, it's using a workaround that will stop working if the new SteamVR action system gets "activated." 

Basically, the new SteamVR plugin doesn't let you read directly from the controllers, instead abstracting it to actions you make with their GUI. The workaround bypasses SteamVR and goes straight to the OpenVR API, which works. But if the action system loads, then a critical API function stops working. 

So the new SteamVR plugin is now an option but you'll have to avoid using the SteamVR prefabs or most newer scripts of theirs.