stereolabs/zed-unity #33: Drifts when using ZED camera in a virtual scene

rkachach, 2018-11-05:
I'm experementing some localization drifts when using the ZED camera in VR scenes. I know that the the camrea is generally meant for AR, but for my use case I need to use the zed-M camera in a VR scene. Going deeper in the code I saw that the code is using the HMD localization but it performs some drift correction upon that:

`
                ar.AdjustTrackingAR(zedPosition, zedOrientation, out r, out v);
`

I think this is the root cause of the driting and I don't understand why this "correction" is performed when the HMD is already providing an accurate localization?

obraun-sl, 2018-11-05:
Hi,

The tracking of the ZED-M and the tracking of the HMD does not have the same origin point (the tracking of the ZED has its origin on the left sensor, and the HMD positions/orientations are provided on the head center. Therefore the function is used to provide the HMD position as if it was taken from the left sensor point of view and then merged into a single transform.

Best in your application would be maybe to disable the ZED tracking when you are in VR only so that you don't have those adjustments mainly made for AR.

rkachach, 2018-11-05:
Hello,

Thank you very much for the quick and detailed response. I followed your recomendation and now the tracking is working properly :)

closed 2018-11-05 by rkachach