stereolabs/zed-unity #40: ZED AR Unity Build for Linux

supsensei, 2018-12-17:
Hi I am running a Unity build using the ZED plugin for Unity. I have done AR passthrough successfully on Windows 10 machine. I have rebuilt the project for Linux 86_64 and it is not able to connect to ZED camera. I know I have the unity setup for the ZED and am using ROS (Robot Operating System) plugin successfully on ROS with the ZED. I want to RUN and AR script with the ZED on Unity and does nto seem to be working. Is this completely unsupported?

Thank you,

supsensei, 2018-12-18:
Just to also note. The examples all work on the Ubuntu 18.04 computer as well as the ZED Explorer, Fu, etc. It just does not connect when I do AR pass through following the tutorial and then doing a unity build for linux.

supsensei, 2018-12-18:
Here is log file using the -logFile option when I run it in unity. SDK is not found. but it is installed and I can run all the examples and ZED apps.
```
Set current directory to /home/initial/Linux_BUILD
Found path: /home/initial/Linux_BUILD/AR_Linux.x86_64
Mono path[0] = '/home/initial/Linux_BUILD/AR_Linux_Data/Managed'
Mono config path = '/home/initial/Linux_BUILD/AR_Linux_Data/Mono/etc'
Preloaded 'ScreenSelector.so'
Preloaded 'libopenvr_api.so'
Display 0 'EV2456 24"': 1920x1200 (primary device).
Desktop is 1920 x 1200 @ 60 Hz
Invalid initial resolution 1 x 1 - forcing to 100 x 100
Initialize engine version: 2018.2.19f1 (06990f28ba00)
GfxDevice: creating device client; threaded=1
Renderer: Quadro P2000/PCIe/SSE2
Vendor:   NVIDIA Corporation
Version:  4.5.0 NVIDIA 415.18
GLES:     0
 GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_AMD_vertex_shader_viewport_index GL_AMD_vertex_shader_layer GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_cl_event GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_compute_shader GL_ARB_compute_variable_group_size GL_ARB_conditional_render_inverted GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_indirect GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_2_compatibility GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_locatio
n GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_fragment_shader_interlock GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gl_spirv GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_gpu_shader_int64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_polygon_offset_clamp GL_ARB_post_depth_coverage GL_ARB_program_interface_quer
y GL_ARB_provoking_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_robustness GL_ARB_sample_locations GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shader_viewport_layer_array GL_ARB_shading_language_420pack GL_ARB_shading_language_include GL_ARB_shading_language_packing GL_ARB_shadow GL_ARB_sparse_buffer GL_ARB_sparse_texture GL_ARB_sparse_texture2 GL_ARB_sparse_texture_clamp GL_ARB_spirv_extensions GL_ARB_stencil_texturing GL_ARB_sync GL_ARB_tessella
tion_shader GL_ARB_texture_barrier GL_ARB_texture_border_clamp GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression GL_ARB_texture_compression_bptc GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_cube_map_array GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_filter_anisotropic GL_ARB_texture_filter_minmax GL_ARB_texture_float GL_ARB_texture_gather GL_ARB_texture_mirror_clamp_to_edge GL_ARB_texture_mirrored_repeat GL_ARB_texture_multisample GL_ARB_texture_non_power_of_two GL_ARB_texture_query_levels GL_ARB_texture_query_lod GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_texture_rgb10_a2ui GL_ARB_texture_stencil8 GL_ARB_texture_storage GL_ARB_texture_storage_multisample GL_ARB_texture_swizzle GL_ARB_texture_view GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_transform_feedback_instanced GL_ARB_transform_feedback_overflo
w_query GL_ARB_transpose_matrix GL_ARB_uniform_buffer_object GL_ARB_vertex_array_bgra GL_ARB_vertex_array_object GL_ARB_vertex_attrib_64bit GL_ARB_vertex_attrib_binding GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_vertex_type_10f_11f_11f_rev GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_ARB_window_pos GL_ATI_draw_buffers GL_ATI_texture_float GL_ATI_texture_mirror_once GL_S3_s3tc GL_EXT_texture_env_add GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend [... cut ...]
(Filename: ./Runtime/Export/Debug.bindings.h Line: 43)

ZED SDK not installed
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 43)

ZED SDK not installed
 
(Filename: ./Runtime/Export/Debug.bindings.h Line: 43)

Setting up 6 worker threads for Enlighten.
  Thread -> id: 7fec25108700 -> priority: 1 
  Thread -> id: 7fec03fff700 -> priority: 1 
  Thread -> id: 7fec037fe700 -> priority: 1 
  Thread -> id: 7febfa31c700 -> priority: 1 
  Thread -> id: 7febf9b1b700 -> priority: 1 
  Thread -> id: 7febf931a700 -> priority: 1 
```

obraun-sl, 2018-12-18:
Hi,
Right now only Windows is supported since the wrapper is provided as a dll library (not compatible under Linux).
For the moment, there is no plan to port the wrapper library under Linux, but we will let you know if this is the case.

supsensei, 2018-12-18:
Thank you for the reply. That is unfortunate. Thank you.

closed 2018-12-18 by supsensei