stereolabs/zed-unity #42: Tracking of ZED_Rig_Stereo is completely not working ( Unity 2018.3, CUDA10, SteamVR v2.2b5 )

dogod621, 2019-01-16:
Hi,

**My Environment:**
- Windows 10
- Unity 2018.3.0f2 (64-bit)
- cuda_10.0.130_411.31_win10
- ZED_SDK_Windows_v2.7.1
- SteamVR Unity Plugin v2.2b5 
- HTC VIVE VR
- ZED MINI
- CPU: Intel Core i5-8500 @ 3.00GHz
- GPU: GEFORCE GTX 1050

**Problem:**
I have been fighting for this bug for a week :-(...
I am very sure that:
- The RGB image captured by of ZED MINI is OK~ 
- The tracking of ZED_Rig_Mono is OK~
- The tracking of  HTC VIVE is OK~ ( I check the value of UnityEngine.XR.InputTracking.GetLocalPosition(UnityEngine.XR.XRNode.Head) and UnityEngine.XR.InputTracking.GetLocalRotation(UnityEngine.XR.XRNode.Head) )

But somehow the tracking of ZED_Rig_Stereo is completely wrong. 
It jumps like crazy, and the tracking is not correct at all. 
Even weird is that I cover the camera of ZED MINI by my hand, 
but the tracking state is always "sl.TRACKING_STATE.TRACKING_OK". 
That means somehow the input data that is used by ZED tracking process is completely wrong, 
and it is not related to camera at all.

**Problem Demo:**
I put a short demo video: https://youtu.be/qUKO7lWh-5g
(I did not change any code in this demo project. I create a new unity project and import SteamVR package and ZED package. I demo the bug by simply add a zed rig into an empty scene.)

[0:00](https://youtu.be/qUKO7lWh-5g) : Tracking of ZED_Rig_Stereo is completely wrong
[0:30](https://youtu.be/qUKO7lWh-5g?t=30) : I cover the camera of ZED MINI by my hand (Always "sl.TRACKING_STATE.TRACKING_OK" @@!!)
[0:55](https://youtu.be/qUKO7lWh-5g?t=55) : Tracking of ZED_Rig_Mono is OK

tony03160, 2019-03-04:
the script not working

[image]
[image]

dogod621, 2019-03-04:
Because I remove the rendering part... @@, I sad it in my post

closed 2022-04-28 by github-actions[bot]