stereolabs/zed-unity #44: Cannot get VIVE tracking at Awake Event, better init ZED camera origin pose at Start Event

dogod621, 2019-01-21:
Hi, 

**My Environment:**
- Windows10
- Unity 2018.3.0f2 (64-bit)
- cuda_10.0.130_411.31_win10
- ZED_SDK_Windows_v2.7.1
- SteamVR Unity Plugin v2.2b5
- HTC VIVE
- ZED MINI
- CPU: Intel Core i5-8500 @ 3.00GHz
- GPU: GEFORCE GTX 1050

I notice the ZEDManager init ZED camera at Awake Event (ZEDManager .cs, Line 832),
I found It will use VIVE tracking to init ZED origin pose  (ZEDManager .cs, Line 1165):
```C#
            ZEDMixedRealityPlugin.Pose pose = arRig.InitTrackingAR();
            OriginPosition = pose.translation;
            OriginRotation = pose.rotation;
            zedRigRoot.localPosition = OriginPosition;
            zedRigRoot.localRotation = OriginRotation;
```

```C#
        public Pose InitTrackingAR()
	{
		Transform tmpHMD = transform;
		tmpHMD.position = UnityEngine.XR.InputTracking.GetLocalPosition(UnityEngine.XR.XRNode.Head);
		tmpHMD.rotation = UnityEngine.XR.InputTracking.GetLocalRotation (UnityEngine.XR.XRNode.Head);

		Quaternion r = Quaternion.identity;
		Vector3 t = Vector3.zero;
		Pose const_offset = new Pose(t, r);
		dllz_drift_corrector_set_calibration_const_offset_transform(ref const_offset);

		zedCamera.ResetTrackingWithOffset(tmpHMD.rotation,tmpHMD.position,HmdToZEDCalibration.rotation,HmdToZEDCalibration.translation);

		return new Pose(tmpHMD.position, tmpHMD.rotation);
	}
```

I test these code without opening new threads (You guys create new thread  to init ZED camera at Awake Event, I just put all of them to Awake Event without new threads.). I found ZED coordinate cannot align with VIVE coordinate, and I found:
```C#
tmpHMD.position = UnityEngine.XR.InputTracking.GetLocalPosition(UnityEngine.XR.XRNode.Head);
tmpHMD.rotation = UnityEngine.XR.InputTracking.GetLocalRotation (UnityEngine.XR.XRNode.Head);
```
Just return identity pose, and then I move those code to Start Event, finally can get correct VIVE pose, and finally align ZED coordinate with VIVE coordinate, so you guys better move init process to Start Event.

obraun-sl, 2019-01-21:
Hi,

If possible, can you make a pull request or provide a diff so that we can check the difference and integrate the changes in the dev branch ?

dogod621, 2019-01-21:
I post my code at https://github.com/stereolabs/zed-unity/issues/42 ...
But my partner use Unity 2018.2 and non-Beta version SteamVR...
His ZED_Rig_Stereo is working correctly....
maybe is my environment cause all these weird problems... QQ...