stereolabs/zed-unity #46: Failed ZED initialization

Airat1995, 2019-02-12:
Hello, thank you for the product but I have a problem with ZED m initialization. When I trying to initialize ZED from Unity (2018.3, 2017.2.20, 2017.2.6), for some reasons, it just crashing without any errors. After full assets reimport, camera initialize without any problems, but only once. Then I must reimport all assets again. I found that program crash inside dllz_open method.
ZED SDK Version: 2.5
CUDA: 9.2
Unity: 2018.3.5f1
OS: Windows 10 x64

obraun-sl, 2019-02-12:
Hi,

Can you update to ZED SDK 2.7 and plugin version 2.7 to see if you still have the crash ?

OB

Airat1995, 2019-02-12:
Hello, sorry for the trouble I created new empty project just with ZED and it's ran without any problem.

closed 2019-02-12 by Airat1995