stereolabs/zed-unity #48: spatial mapping and .area file is not exist

478161648, 2019-03-01:
when I stop the spatial mapping,there is only a ZEDMesh.obj, the .area file is not exist.
is someone know this problem? and what should i do ,if i want use the .area file.

Abrahamh08, 2019-03-07:
Same problem for me. They used to generate, now they don't. No idea why.

ckorris, 2019-03-08:
Hi all, I came across this recently. The issue is that spatial memory needs to be enabled in ZEDManager in order for it to be recorded. But at some point one of us must've unchecked spatial memory (or updated the prefab) and it got left that way when we uploaded it. Sorry about that. 

For now, just go to the ZEDManager's Inspector in ZED_Rig_Stereo and enable Spatial Memory. That should fix it, and we'll make it default to enabled in the next update.

478161648, 2019-03-08:
The stereolab's answer:

Chris (Stereolabs)
Mar 1, 09:28 PST
Hello, I was able to reproduce your issue immediately. The problem is that the ZEDManager component in ZED_Rig_Stereo does not have Spatial Memory enabled. You can enable it in its Inspector. You should then have an .area file after you scan. I will fix this for our next patch, as it should be enabled by default. My apologies. 

Chris
Stereolabs Support
--

ckorris, 2019-03-08:
^ That's me, and that's how I found out about the issue. Thanks for reporting it and sorry I missed that you also posted on GitHub.

closed 2022-04-28 by github-actions[bot]