stereolabs/zed-unity #50: Support for HDR pipeline

otite, 2019-03-23:
Hi, it would be awesome if the unity plugin could support the HDR render pipeline of unity.
G.

ckorris, 2019-03-29:
Thanks for the suggestion. A few people have asked for this, though we've researched it and it will be a long and complicated process to add full HDRP support. It _is_ on the To Do list, but as the vast majority of users are using Standard, please note it'll still be a little while as we've prioritized other features.

djesko1, 2019-03-29:
Was going to ask this also. It will be essential for my project going forward.

flaming910, 2019-10-10:
Could you maybe give some guidance on what could be changed to use the ZED with HDRP?

peterfiftyfour, 2021-01-23:
Hey all, any update on this? I managed to get some stuff partially working. The only issue was in LookAtCameraPartialAxis where RenderPipeline.beginCameraRendering += LookAtCamera; should be RenderPipelineManager.beginCameraRendering - would love to use some fancy vfx stuff! Cheers

davidezhang, 2021-11-12:
Would love to hear any updates as well! @peterfiftyfour when I changed it to RenderPipelineManager.beginCameraRendering, there is this error: No overload for 'LookAtCamera' matches delegate 'Action<ScriptableRenderContext, Camera>'. Would appreciate any pointers!

closed 2022-04-27 by github-actions[bot]