stereolabs/zed-unity #51: Point Cloud

tomitrescak, 2019-03-29:
Thanks for the amazing set of tools. Previously there was a demo scene which visualized point cloud from camera. I can’t find a way how to do it anymore.

I have a bit of reversed scenario when I want to visualize point clouds in ve fed from Zed camera. Thanks for any help.

ckorris, 2019-03-29:
That sample was from quite awhile ago, and we removed it because it was rather plain once we added the Planetarium, Drone Battle, etc. But the script you need is still there. 

Just make a scene with the ZED_Rig_Mono prefab in it somewhere, and another GameObject with ZEDPointCloudManager. When you hit Play, you should see a point cloud in front of the second GameObject. 

To hide the ZED's normal output from the scene, disable the camera in ZED_Rig_Mono's child, Left_Camera. (ZEDPointCloudManager only needs ZEDManager to work, which is at the root object in ZED_Rig_Mono, but there's currently an issue where ZEDManager outside a rig throws an exception.)

tomitrescak, 2019-04-01:
Thanks, works like a charm!

closed 2019-04-01 by tomitrescak