stereolabs/zed-unity #58: 'CAMERA_NOT_DETECTED' after retreiving texture

Rosthouse, 2019-04-26:
Hello again

Last week I reported issue #53, which was solved with the new 2.8 release. After adapting my code to the new release, I now face another issue. After the texture is retrieved for the first time, the method 'camera.UpdateMesh(...)' will always report a status of 'CAMERA_NOT_DETECTED'.

I tested various different methods to structure the calls so that this issue would go away, but nothing really worked.

This is the log output I'm getting (cleaned up a little, removing local paths and such):

```
Reloading assemblies for play mode.
Begin MonoManager ReloadAssembly
Initializing Extension Manager v2019.1.0 for Unity v2019.1.0f2
Registering platform support modules:
Registered platform support modules in: 0.0401148s.
Native extension for WindowsStandalone target not found
Refreshing native plugins compatible for Editor in 0.32 ms, found 1 plugins.
Preloading 0 native plugins for Editor in 0.00 ms.
Mono: successfully reloaded assembly
- Completed reload, in  1.278 seconds
Platform modules already initialized, skipping
Load scene 'Temp/__Backupscenes/0.backup' time: 0.376538 ms
Refreshing native plugins compatible for Editor in 0.31 ms, found 1 plugins.
Warning: Screen size should be set to 16:9 aspect ratio
UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
<InitZED>d__193:MoveNext() (at Assets\ZED\SDK\Helpers\Scripts\ZEDManager.cs:1277)
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr) (at C:\buildslave\unity\build\Runtime\Export\Scripting\Coroutines.cs:17)

(Filename: Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs Line: 1277)

ZED (Spatial mapping) >> WARNING: Trying to allocate 4096 Mb, but only 1275 Mb available. Reallocating 1024 Mb
Iteration 0
UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
<UpdateMesh>d__17:MoveNext() (at Assets\_Project\Scripts\Mapping\ProgressiveMapper.cs:95)
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr) (at C:\buildslave\unity\build\Runtime\Export\Scripting\Coroutines.cs:17)
UnityEngine.MonoBehaviour:StartCoroutineManaged2(IEnumerator)
UnityEngine.MonoBehaviour:StartCoroutine(IEnumerator) (at C:\buildslave\unity\build\Runtime\Export\Scripting\MonoBehaviour.bindings.cs:91)
ProgressiveMapperBehaviour:StartMapping() (at Assets\_Project\Scripts\Behaviours\ProgressiveMapperBehaviour.cs:53)
ProgressiveMapperBehaviour:ToggleMapping() (at Assets\_Project\Scripts\Behaviours\ProgressiveMapperBehaviour.cs:88)
UnityEngine.Events.InvokableCall:Invoke() (at C:\buildslave\unity\build\Runtime\Export\UnityEvent\UnityEvent.cs:166)
UnityEngine.Events.UnityEvent:Invoke() (at C:\buildslave\unity\build\Runtime\Export\UnityEvent\UnityEvent\UnityEvent_0.cs:58)
UnityEngine.UI.Button:Press() (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\UI\Core\Button.cs:66)
UnityEngine.UI.Button:OnPointerClick(PointerEventData) (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\UI\Core\Button.cs:108)
UnityEngine.EventSystems.ExecuteEvents:Execute(IPointerClickHandler, BaseEventData) (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\ExecuteEvents.cs:50)
UnityEngine.EventSystems.ExecuteEvents:Execute(GameObject, BaseEventData, EventFunction`1) (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\ExecuteEvents.cs:261)
UnityEngine.EventSystems.StandaloneInputModule:ReleaseMouse(PointerEventData, GameObject) (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\InputModules\StandaloneInputModule.cs:195)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMousePress(MouseButtonEventData) (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\InputModules\StandaloneInputModule.cs:644)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent(Int32) (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\InputModules\StandaloneInputModule.cs:552)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent() (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\InputModules\StandaloneInputModule.cs:532)
UnityEngine.EventSystems.StandaloneInputModule:Process() (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\InputModules\StandaloneInputModule.cs:289)
UnityEngine.EventSystems.EventSystem:Update() (at C:\buildslave\unity\build\Extensions\guisystem\UnityEngine.UI\EventSystem\EventSystem.cs:377)

(Filename: Assets/_Project/Scripts/Mapping/ProgressiveMapper.cs Line: 95)

ZED (Mesh) >> ERROR: No texture images found.
Hints :
        -applyTexture() doesn't work for loaded Mesh.
        -SpatialMappingParameters::save_texture needs to be true.
Started spatial mapping
UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
ProgressiveMapperBehaviour:StartMapping() (at Assets\_Project\Scripts\Behaviours\ProgressiveMapperBehaviour.cs:54)
ProgressiveMapperBehaviour:ToggleMapping() (at Assets\_Project\Scripts\Behaviours\ProgressiveMapperBehaviour.cs:88)
[... 128 lines cut ...]
2019-04-26.13:44:55:	:	ENTER requestMeshAsync
2019-04-26.13:44:55:	:	EXIT requestMeshAsync
2019-04-26.13:44:55:	:	ENTER updateMesh
2019-04-26.13:44:55:	:	FUNC updateMesh : call getMeshRequestStatusAsync success
2019-04-26.13:44:55:	:	FUNC updateMesh : call retrieveMeshAsync success : 0
2019-04-26.13:44:55:	:	FUNC updateMesh : fill vertices 0 0
2019-04-26.13:44:55:	:	ENTER requestMeshAsync
2019-04-26.13:44:55:	:	EXIT requestMeshAsync
2019-04-26.13:44:56:	:	ENTER updateMesh
2019-04-26.13:44:56:	:	FUNC updateMesh : call getMeshRequestStatusAsync success
2019-04-26.13:44:56:	:	FUNC updateMesh : call retrieveMeshAsync success : 0
2019-04-26.13:44:56:	:	FUNC updateMesh : fill vertices 0 0
2019-04-26.13:44:56:	:	ENTER requestMeshAsync
2019-04-26.13:44:56:	:	EXIT requestMeshAsync
2019-04-26.13:44:57:	:	ENTER updateMesh
2019-04-26.13:44:57:	:	FUNC updateMesh : call getMeshRequestStatusAsync success
2019-04-26.13:44:57:	:	FUNC updateMesh : call retrieveMeshAsync success : 58
2019-04-26.13:44:57:	:	FUNC updateMesh : fill vertices 58 3474
2019-04-26.13:44:58:	:	ENTER requestMeshAsync
2019-04-26.13:44:58:	:	EXIT requestMeshAsync
```

In iteration 2, the camera first returns a texture, which I am able to overlay over the generated mesh. This works great! However in the following Iteration (Iteration 3), the camera can apparently not be detected by Unity anymore ('CAMERA_NOT_DETECTED'). Although, I still have a live-feed inside Unity from the camera, which I would assume should not be possible anymore.

I published the code that is creating this error [here](https://gist.github.com/Rosthouse/674da3f25370e0c36468f83184d1e305) if anyone wants to have a look at it :)

ckorris, 2019-04-29:
Try replacing the sl_unitywrapper.dll in your ZED/SDK/Plugins/win64 folder with this: https://drive.google.com/open?id=1YXvl5Umzt3djBTZni3-P2NyYRbBCeMae

In the wrapper, both UpdateMesh and RetrieveMesh don't continue if a flag called "isTextured" is turned on. Normally, this happens the first time you retrieve the mesh when a texture is ready. This was because by design, you normally only retrieve the texture once in Unity. 

The fact that it says it can't connect to the camera is due to a mistake: the wrapper checks that the camera has been connected AND that it's not yet textured in the same if statement, and returns the CAMERA_NOT_CONNECTED error if that if statement evaluates to false for either reason. In your case, the camera is still connected but it thinks it should stop because it's already been textured. 

In the .dll I sent, I simply removed the check for the isTextured flag. The errors you described go away. 

After running it, I don't see any more error messages, though it doesn't appear to be properly adding the new textures to the mesh. I'll toss the ball back to your court to figure out why, since it could be related to your progressive mesh setup, but please let me know if it turns out to be related to the plugin.

Rosthouse, 2019-04-30:
Thanks for the dll, can confirm that the error does **not** pop up anymore. Would it make sense to pass a boolean to the API defining wheter to check for the `isTextured` flag or not, defaulting to current behaviour? Like this, it wouldn't impact previous implementations.

I tested the texturing and while I do see some glitches, they mostly stem from moving the camera too fast. If I hold it steady it is mostly spot on. Maybe the transform of the camera is calculated before the UVs are generated and thus the mapping isn't correct anymore? I'll do some more testing on this.

Anyway again, top job :)

ckorris, 2019-04-30:
Glad it helped!

In your case, it would not be necessary to pass in the isTextured flag. In the wrapper, that flag is to indicate that it has _retrieved_ a texture from the mesh, not just that the spatial mapping module is set to texture the mesh. Currently, we only use that to stop functions from working once we've retrieved that texture. 

So are you currently able to do what you intend? If you move the camera around slowly, do multiple chunks in different positions get new texture updates? That's the part I couldn't make happen, but I was also moving the camera a lot.

Rosthouse, 2019-05-06:
sorry for replying so late. 

**Yes, currently I am able to do what I'm intending**. I'm currently testing different configurations for getting the optimal performance, which will take some time. I'm also thinking if it would help if methods like ApplyTexture on the camera could provide an async variation, which would prevent the application from freezing for a few frames. As said, preformance is not my main concern right now, but it would be nice for another day, I suppose.

Rosthouse, 2019-05-16:
Alright, I tested this some more. And I (think) I found a solution to the "fragmenting" problem. 

I guessed the problem is that while the camera is applying the texture Unity is completely frozen and can't update the position of the ZED camera. As soon as `ApplyTexture` returns, the positions are out of sync and the chunks are placed who-knows-where. So I figured I'd conjure a way to let Unity continue executing while the camera is applying the texture. But since some key camera operations need to happen on the main thread (like creating textures) I'm not able to use threads to do this. Instead I leveraged coroutines. My idea was then to yield execution of the coroutine while the camera is working.

Since `ApplyTextures` can run outside of the main thread, I figured I'd use a [Task](https://docs.microsoft.com/de-de/dotnet/api/system.threading.tasks.task?view=netframework-4.8) to create a lightweight thread, that allows me to check if it is finished. Then I needed a way to let Unity know to continue working while this Task is not finished. I created a new [YieldInstruction](https://docs.unity3d.com/ScriptReference/CustomYieldInstruction.html), which allows me to yield until the given task is completed. You can view that [here](https://gist.github.com/Rosthouse/c58048def0b4915d752d83f46e0e1666)

After integrating this change into the [`ProgressiveMapper`](https://gist.github.com/Rosthouse/674da3f25370e0c36468f83184d1e305#file-progressivemapper-cs-L101) the whole thing just immediately started working. I can move the camera anywhere and the texture & mesh are always correct.  Also now I don't experience any lagging (obviously) and always have full control over my unity instance.

This is awesome. I consider this case closed, if @ckorris doesn't have any more to add to this :)

ckorris, 2019-05-16:
Glad it's working well! I don't foresee any issues with what you did, and appreciate you sharing the details + code. 

And yes, I could rant about Unity forcing you to make textures on the main thread, but my cardiologist forbade it. The worst I've experienced is EncodeToPNG, which'll lock up the main thread for 200-500ms for a decently-sized image.

If you ever take a video of your real-time scanning in action, I'd love to see it!

Rosthouse, 2019-05-16:
I'll see if I can make a video that :)

closed 2019-05-16 by Rosthouse