stereolabs/zed-unity #60: ZED_Spatial_Mapping missing script

alicehaupais, 2019-05-09:
Hello,

I would like to report a new issue regarding the new version v2.8 of the ZED SDK.

I wanted to do the tutorial about Spatial Mapping : https://www.stereolabs.com/docs/unity/spatial-mapping-unity/

However, the prefab mentionned ZED_Spatial_Mapping is missing a script : ZEDSpatialMappingManager.cs. I looked for another script and I found ZEDSpatialMapping.cs but this one contains compilation errors.

[image]

[image]

I tried also to run the tutorial 5 - Spatial Mapping but I have also an execution error.
[image]

Could you tell me if there is something new with the setup of the Spatial Mapping Feature ?

Thank you for your answer !

alicehaupais, 2019-05-09:
I successfully managed to make the Spatial Mapping script worked !

The class was not deriving from MonoBehaviour ... (script contains also this mistake in the StereoLabs git ....)

Start Spatial Mapping is now working but is not able to record anything ...

obraun-sl, 2019-05-10:
Please note that the spatial mapping module is now directly in ZEDManager (ie ZED_Rig_Mono or Stereo prefab) and it's not a standalone module. 
Basically you just need to remove the spatial mapping prefab and replace with a ZED Rig prefab.

Take a look at the spatial mapping scene.

alicehaupais, 2019-05-10:
Yes I managed to make the spatial mapping work from Unity. It may be time to update the tutorials ....

closed 2019-05-10 by alicehaupais