stereolabs/zed-unity #61: Access ZED_unity raw images?

hankhhub, 2019-05-14:
Hi, I would like to know if it is possible to run the zed camera in unity game and access raw images for further processing at the same time?

ckorris, 2019-05-14:
Sure, that's possible. You would have to grab those raw images from within Unity, as only one connection to the camera can be active at the same time (unless you want to use our streaming module to broadcast the images to another instance of the SDK elsewhere). 

To do that, call [ZEDCamera.CreateTextureImageType](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/NativeInterface/ZEDCamera.cs#L1329) with VIEW.LEFT_UNRECTIFIED or VIEW.RIGHT_UNRECTIFIED as the first argument. This will make a Texture2D that is automatically updated by our wrapper as it runs, and will not conflict with anything else you're doing in Unity with the ZED. 

Then you can do whatever you need with those images for your 'further processing,' including send them to a different application. Just keep in mind that the data will be on the GPU as it's a Texture2D, so you'll need to marshal over the data to access it on the CPU.

hankhhub, 2019-05-15:
By further processing, I meant to send images through socket to another computer and process it with libraries like openCV. Nonetheless, thank you for the quick response!

ckorris, 2019-05-15:
Ah, okay. Then you have two options: 

1. Do like I said above, then send the images to the other computer via any protocol you'd like. 

2. Use our new network streaming feature to broadcast the ZED's input to any local computer on the network. You can do that in ZED SDK/Unity plugin 2.8 from ZEDManager's Streaming section. Just start hosting, then have the other computer connect. 

#2 would be a lot simpler - the only catch is that the other computer also has to be compatible with the ZED. But it doesn't have to be Unity - it could be an app built on our C++ API or any other plugin we offer that supports the new streaming feature. In fact, for OpenCV, you can build a C++ or Python app with [this sample](https://github.com/stereolabs/zed-opencv), and modify the camera's initialization to take a stream instead of a direct USB input, as described [here in our docs](https://www.stereolabs.com/docs/video/using-video/#streaming-a-video). 

Does that help?

hankhhub, 2019-05-16:
This is awesome, it's what I was looking for! However, I do have one more minor problem, when I moved the project to 2.8, the scripting define symbols ZED_OCULUS;ZED_STEAM_VR keeps refreshing each time I save in unity editor. Since I am not using SteamVR, unity is not allowing me to compile my code.

ckorris, 2019-05-16:
Ah, that happens with newer versions of Unity because they now include an OpenVR package file automatically. Before, we were looking for a file named "OpenVR" to decide if the SteamVR package had been imported. We patched the change into the source code on Git but it's not reflected in the 2.8 release download. 

See this line: https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/Interactions/ZEDControllerTracker.cs#L692

Yours probably looks for "OpenVR" instead of "SteamVR_Camera". Make the change (or replace the whole script) and you should be good!

hankhhub, 2019-05-17:
The streaming works great. However, one more concern I have is whether the Python API or Unity Plugin is capable of sending other data types asides from ZED stream? For example, I want to receive ZED image then process it and send back image vector coordinates.

ckorris, 2019-05-21:
This is not a direct feature of our Python wrapper or Unity plugin, but you could certainly add your own code to send that data over the network. Unity has a number of networking options since it's made for multiplayer games, though a 3rd party library from the Asset Store may be better since you're sending data outside Unity.

hankhhub, 2019-05-23:
So I tried your recommendation of accessing raw ZED images and marshal it over to CPU. I wanted to test if that is functioning by saving it to a png file, but what I got was a 16kb small png file of complete white image. Is there something wrong with the way I write the bytes?

using UnityEngine;
using sl;
using System.Runtime.InteropServices;
using System;
using System.IO;

public class ZEDTest : MonoBehaviour
{
    public GameObject ZED_Rig_Stereo;
    private ZEDManager zedmanager;
    sl.ERROR_CODE err = sl.ERROR_CODE.CAMERA_NOT_DETECTED;
    
    // Start is called before the first frame update
    void Start()
    {
        zedmanager = ZED_Rig_Stereo.GetComponent<ZEDManager>();
    }

    // Update is called once per frame
    void Update()
    {
        // Retrieve ZEDCamera image from GPU for access in CPU
        ZEDMat zedmat = new ZEDMat(1280, 720, ZEDMat.MAT_TYPE.MAT_8U_C4, ZEDMat.MEM.MEM_CPU); //Alternative Matrix Type: MAT_8U_C4           
        while (err != sl.ERROR_CODE.SUCCESS)
        {
            err = zedmanager.zedCamera.RetrieveImage(zedmat, VIEW.LEFT_UNRECTIFIED, ZEDMat.MEM.MEM_CPU);
            Debug.Log(err);
        }
        IntPtr matptr = zedmat.GetPtr(ZEDMat.MEM.MEM_CPU);
        
        int matwidth = zedmat.GetWidth();
        int matheight = zedmat.GetHeight();
        int len = 1280 * 720 * 4;
        byte[] frame_bytes = new byte[len];
        Marshal.Copy(matptr, frame_bytes, 0, len); 
        
        if (frame_bytes != null)
        {                     
            Texture2D image = new Texture2D(1280, 720, TextureFormat.RGBA32, false);
            image.LoadRawTextureData(frame_bytes);
            byte[] bytes = image.EncodeToPNG();
            Destroy(image);
            File.WriteAllBytes(Application.dataPath + "/../zed.png", bytes);
            Application.Quit();
        }
    }
}

ckorris, 2019-05-23:
When I put that code into my game, it works. However, note that: 

1. You can't get a texture from the ZED until the ZED is finished initializing. You can use ZEDManager.IsZEDReady to check this, or subscribe to the ZEDManager.OnZEDReady event. 
2. EncodeToPNG is a _very_ slow operation, often measured in the hundreds of milliseconds. With that in Update(), you're grinding the main thread down to a halt. Add to that the fact that the file path doesn't change and you're constantly reading to the same place, and it wouldn't surprise me if something goes wrong along the way. Usually for testing this kind of thing, I'd suggest performing the save with a key command, like:

```
if(Input.GetKeyDown(KeyCode.Space)
{
    //Save .png
}
```

I'd try that and see how it goes. 

A few other notes for things you'll likely face soon: 

- The texture format is BGRA by default, so if you set the TextureFormat to RGBA32 like you did, the image color will be off (people will look blue, usually). Try BGRA32. 
- The image will be upside down. For testing, I just flip the bytes manually with code like this: 

```
byte[] flippedbytes = new byte[matwidth * matheight * 4];
int steplength = matwidth * 4; //Also retrievable from ZEDMat.GetStepBytes();
for(int i = 0; i < matbytes.Length; i += steplength)
{
       Array.Copy(matbytes, i, flippedbytes, flippedbytes.Length - i - steplength, steplength);
}
```

Obviously that's not too performant but at least it validates the copying method. 
* If you indeed need to save images repeatedly, Unity will unfortunately not let you call EncodeToPNG on a separate thread for some reason. But you can use a third party library to do this. I once used MagickImage to do this instead, and queued up images in real-time to be saved in not-real-time on a separate thread. The encoding is a lot slower with that library (I didn't thoroughly explore all possible libraries) but there's a good tutorial for getting it to work in Unity [here](https://www.google.com/search?q=magickimage+unity&oq=magickimage+unity&aqs=chrome..69i57.2844j0j7&sourceid=chrome&ie=UTF-8).

hankhhub, 2019-05-24:
I found the mistake I made in copying method in 
while (err != sl.ERROR_CODE.SUCCESS)
    {
        err = zedmanager.zedCamera.RetrieveImage(zedmat, VIEW.LEFT_UNRECTIFIED, ZEDMat.MEM.MEM_CPU);
        Debug.Log(err);
    }
where once I get a SUCCESS, my program will never grab an image again, which is why my png are empty. Also, changing TextureFormat to BGRA32 did fix the blue-ish issue, big thanks!

I am now trying to implement sending the ZED video stream through the network in my own code. I can successfully receive the image in my Python program. But the speed is just not comparable to that of ZED API. I couldn't achieve a 720p stream in real time like the ZED samples you mentioned earlier. I suspect using the a codec is necessary to reduce the data size. Do you have any suggestions or workarounds on this?

ckorris, 2019-05-28:
Ah! That makes sense for the issue you had. Sorry I missed that. 

For recommendations, in the past, I used EncodeToJpg, which is much much faster than EncodeToPng, can make much smaller files and lets you specify a quality level, but is also lossy compression. I also sent at lower FPS, like 10-15 at 720p resolution, for which you normally get frames at 60FPS. This made it pretty easy on the network, but of course reduced the quality of the video feed significantly. Would that work for your use case, or is it necessary to maintain the same quality? 

Also, just to be clear, you do _not_ intend on using the ZED SDK to process these images in the other application, correct? I ask because the ZED SDK will not accept custom image inputs - it has to come from a USB connection, a recorded SVO, or a network stream using our own network features. This is a restriction we unfortunately have to put in place to protect our IP - I hope you understand. But if you're doing your own image processing, then no problem :)

hankhhub, 2019-05-28:
Maintaining the quality isn't my priority, so I think I can give it a try. However, since I am doing the image copying and network stream in a subthread, Unity most likely won't let me call  EncodeToJPG.

I am not sure what you meant by custom image inputs, but basically I'm just trying build a ZED USB stream to Python application, except I am implementing my own network protocol and image processing in Python. I hope I'm not violating anything :)

ckorris, 2019-05-28:
Ah, yes, it probably won't. Though if you use a third party library in that subthread to encode to jpg, it'll likely be very fast. 

Regarding the custom inputs, it sounds like you're good! :) I was referring to if you were trying to give images you created yourself (from the network or otherwise) to the ZED SDK to process depth, tracking, etc. There is no way to make this work outside of our own protocols, because then otherwise people could use our (free) SDK with any camera they wanted.

hankhhub, 2019-05-29:
Ah, understood :D! 
So now I found a new issue when I am testing for an alternative way to stop the Python program asides from detecting a key. I wanted to detect if streaming from ZED Unity stops by counting number of ERROR_CODE.NOT_A_NEW_FRAME and automatically exit the Python routine. However, as a result, I couldn't even get one frame from the stream and program exits. Moreover, the Unity editor freezes on game exit as well. I'm posting my script below, it's based on the ZED opencv example. Do you have a clue on whats wrong?

key = ' '
    idle_frames = 0
    idle_threshold = 1000
    while key != 113:
        err = zed.grab(runtime)

        if err == sl.ERROR_CODE.SUCCESS :
            idle_frames = 0
            # Retrieve the left image, depth image in the half-resolution
            zed.retrieve_image(image_zed, sl.VIEW.VIEW_LEFT, sl.MEM.MEM_CPU, int(new_width), int(new_height))
      
            image_ocv = image_zed.get_data()                      
            cv2.imshow("Image", image_ocv)     
            key = cv2.waitKey(10)

            process_key_event(zed, key)
        else:
            idle_frames += 1
        print(idle_frames)    
        if idle_frames > idle_threshold:
            print("Camera Idle Timeout")
            break       
    cv2.destroyAllWindows()
    zed.close()

closed 2022-04-27 by github-actions[bot]