stereolabs/zed-unity #62: Object becomes invisible in a game tab when "interaction behavior" script added

khj4396, 2019-05-17:
I am using Oculus VR and ZED mini.
When I add the C# script from an example "interaction behavior," to the game object, the object becomes invisible when the game is playing. I can still see the object in the headset though. Please see the attached images. 

Any help will be greatly appreciated! 
 
[image]
[image]

ckorris, 2019-05-20:
Ah, I see you're using the Leap Motion plugin. Their Interaction Engine is a bit complicated, but I suspect it has to do with layers. A few questions to help me solve it: 

- Does this happen if you aren't using the ZED Mini, and just a regular Unity camera and/or the Leap Motion rig? I want to make sure it's because of how it's interacting with our plugin. 
- Does the object's layer change when you pick it up? 
- Are you using the ZED 2.8 plugin? (Older versions were dependent on layers to hide certain objects but 2.8 is not)
- I believe you can change the "Interaction Layer" for the Leap Motion plugin in settings. If that's correct, does changing that layer have an effect on this problem? 
- If that's correct that there is an Interaction Layer that you can change, can you make sure the editor camera is set to see that? You can check this by clicking the "Layers" dropdown near the very top right (next to the Layout one) and making sure the eye icon next to the interaction layer is open. You may need to rename that layer in order for it to be visible.

khj4396, 2019-05-28:
Thank you so much for your prompt reply and detailed explanation! It was layer that causes the issue. I changed the layer setting and it works! thank you very much,

vutran000, 2019-12-10:
I am also using the Leap Motion plugin with the Zed Mini and am experiencing the same problem. I also made sure my layers were all visible, but it didn't fix the problem.

ckorris, 2019-12-10:
@vutran000 did you try the other suggestions? Namely, changing the interaction layer in Leap Motion, if that's an option?

khj4396, 2019-12-10:
FYI, I changed the interaction layer as suggested and it worked for me.

vutran000, 2019-12-10:
I added 3 new layers for use with the Leap Interaction Manager, instead of using their auto generate layer option. It looks to me that the ordering of the layers matter. Objects on the User Layer 8 will only render on the left eye but not the right eye. While objects in the User Layer 9 will not render on either eye.

ckorris, 2019-12-10:
That's interesting. Are you using an older version of the ZED plugin? We used to use layers 8 and 10 for hiding the left and right image canvases from the opposite camera, which would explain your issue exactly. But we later moved them to higher layers (28 and 30 I believe) and then removed that layer system altogether in favor of a rendering event callback.

vutran000, 2019-12-10:
I'm currently using 2.8.2

vutran000, 2019-12-10:
I can get around the problem by using any of the other builtin layers instead, but that causes errors to come up from Leap Motion not wanting to share its layers with other objects.

ckorris, 2019-12-10:
That makes me think the Leap Motion plugin is doing something unique with how it's rendering those objects. but it's expecting a specific kind of camera setup which is different than ours. 

I'd first change those layer masks to something higher, like 20+. I'd also check to see if the culling masks of the cameras on the ZED stereo rig change at runtime, or when you pick something up.

ckorris, 2019-12-10:
I just looked at the prefabs, and it looks like they still have some layers hidden by default, left over from a long time ago. 

Now that you've named those new layers, you should be able to go to the Left_eye and Right_eye objects, open their Culling Mask dropdown, and enable all the relevant layers. 

Let me know if that fixes the issue.

vutran000, 2019-12-10:
That fixed it!
Thank you!!

ckorris, 2019-12-10:
Thanks for reporting! I've corrected it in an internal build and it'll be fixed in our next release.

closed 2022-04-27 by github-actions[bot]