stereolabs/zed-unity #63: ZED v2.8 remote streaming not working

hankhhub, 2019-06-03:
I have been testing with sending ZED Unity stream and receiving the stream with zed opencv python. The application works just fine when I am running on the same computer. However, once I try to send stream to a remote python server, I get the error "Camera Not Detected". My python ZED camera initialization is like this:
 # Create a ZED camera object
    zed = sl.Camera()

    # Set configuration parameters
    init = sl.InitParameters()
    init.set_from_stream("my IP address", 30000) 
    
    # Open the camera
    status = zed.open(init)
    if (status != sl.ERROR_CODE.SUCCESS):
        print(repr(status))
        zed.close()
        exit(1)

ckorris, 2019-06-04:
Is the Python server on the same local network, or is it going across the Internet? At the moment, I'm afraid our streaming module only supports local networks.

ckorris, 2019-06-04:
Actually, you might be able to make it work over the Internet by forwarding the port on the server that's one above the port you chose, in addition to the port itself. For example, try forwarding port 30000 and 30001. I'd recommend trying both TCP and UDP.

hankhhub, 2019-06-04:
Both the Unity and Python applications are connected to same ethernet network. I am confused, because I can send simple text with my socket program to the other machine just fine. I tried opening windows firewall ports with TCP and UDP and that didn't work. I also tried ports other than 30000, and I will get "NO GPU Compatible"

ckorris, 2019-06-04:
> I also tried ports other than 30000, and I will get "NO GPU Compatible"

Oh, hang on, is the Python server compatible with the ZED SDK? With a CUDA-capable NVIDIA GPU? I remember from your other posts that you were just going to do image processing on the Python server.

If it's not, then I suspect when you're trying the other ports and get the GPU error, that it's actually finding the camera just fine. But our streaming module is designed to send images from one ZED SDK to another ZED SDK. This is useful for a few cases, like when you have many low-power embedded devices with ZEDs sending the signal to a central server for processing. You can't really use the ZED SDK without being CUDA compatible. 

If that's the case, then you'll have to apply the images to regular textures in Unity and then send them over via a different protocol.

hankhhub, 2019-06-04:
Perhaps I confused you, both my machines have CUDA-capable GPU and have ZED SDK installed. My python server is built from ZED plugin, exactly the one from ZED python streaming examples. The one difference I see now is one machines has v2.8.0 and the other v2.8.1 SDK. 

Also I tried sending and receiving with ZED SDK on the same computer with different ports. Port 30001 will give me NO GPU Compatible but some ports like port 80 works fine.

obraun-sl, 2019-06-06:
Hi ,

On the sender side, the port must be a even number such as 80 or 30000. The port+1 is used for Control Data then. However, the error should not be NO_GPU_COMPATIBLE. Do you have this error with Unity or Python plugin ?

hankhhub, 2019-06-06:
I have NO_GPU_COMPATIBLE only in Python plugin when using odd number ports. Even number ports works fine receiving and sending on the same computer, but gives CAMERA_NOT_DETECTED when receiving on remote machine (connected in local network)

obraun-sl, 2019-06-07:
Did you double-check that "my Ip Address" in your code is indeed the sender address ?

hankhhub, 2019-06-07:
Ok, so I tried the ZED Python API camera_streaming examples and tested sl.Camera().get_device_list(). On the first machine, running get_device_list() on the receiver returns "ZED-M (0) \\?\usb#vid_2b03&pid_f682&mi_00#6&2bace81e&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global SN10027917 AVAILABLE_". Running the other machine as receiver returns "ZED-M (0) /dev/video0 SN10027481 AVAILABLE". So I think I do have one sender on the network each time. 

One computer installed ZED SDK v2.8.0 and the other v2.8.1. The receiving works for v2.8.0 after downloading calibration file "ZED (Init) >> No calibration file found. Downloading it...
calibration file downloaded." However, v2.8.1 receiver will show "ZED (Init) >> ERROR: Executable was designed for SDK v2.8.0 / installed v2.8.1" then followed by CAMERA_NOT_DETECTED

closed 2022-04-27 by github-actions[bot]