stereolabs/zed-unity #64: ZEDControllerTracker issues with SDK 2.8 and Unity 2018.4

derzorngottes, 2019-06-05:
I upgraded to Unity 2018.4 because it is LTS and increases stability for the Oculus Integration.

While trying to add the ZED plugin version 2.8, I get the following errors. First, it gives me the warning about obsolete APIs that need to be upgraded. After upgrading is complete I see this output:

> Assets\ZED\SDK\Helpers\Scripts\Interactions\ZEDControllerTracker.cs(5,7): error CS0246: The type or namespace name 'Valve' could not be found (are you missing a using directive or an assembly reference?)
> 
> Assets\ZED\SDK\Helpers\Scripts\Interactions\ZEDControllerTracker.cs(325,29): error CS0246: The type or namespace name 'TrackedDevicePose_t' could not be found (are you missing a using directive or an assembly reference?)
> 
> Assets\ZED\SDK\Helpers\Scripts\Interactions\ZEDControllerTracker.cs(396,33): error CS0246: The type or namespace name 'EVRButtonId' could not be found (are you missing a using directive or an assembly reference?)
> 
> Assets\ZED\SDK\Helpers\Scripts\Interactions\ZEDControllerTracker.cs(36,5): error CS0246: The type or namespace name 'VRControllerState_t' could not be found (are you missing a using directive or an assembly reference?)
> 
> Assets\ZED\SDK\Helpers\Scripts\Interactions\ZEDControllerTracker.cs(52,20): error CS0103: The name 'OpenVR' does not exist in the current context

This happens for 2.7 or 2.8. If I downgrade Unity to 2018.2.11 it works, but then the Oculus integration is broken and requires a workaround to fix.

derzorngottes, 2019-06-05:
Update: identified the cause of the issue is two checks that are made in the ZEDControllerTracker file:

> if (CheckPackageExists("OpenVR"))

Which erroneously identifies a script named "openvr_api" in the new Oculus Integration as the OpenVR package and sets the ZED_STEAM_VR symbol. 

Changing this line to obfuscate the name fixed the issue:

> if (CheckPackageExists("N_OpenVR"))

ckorris, 2019-06-05:
That's correct, and the same issue was presenting itself in Unity 2019 if you had the OpenVR package in your Packages folder. 

We actually [applied a fix already](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/Interactions/ZEDControllerTracker.cs#L692), same as you but with "SteamVR_Camera" instead of "N_OpenVR", though that change came after we uploaded to the Release page, so you'd have to pull the source code to have it applied. 

We'll be releasing a 2.8.1 version of the plugin soon which includes this fix. 

I'm sorry you had the issue, and thanks for reporting the solution.

closed 2019-06-05 by ckorris

foungyu, 2019-08-30:
Is this fixed in 2.8.1? I still have issue with 2.8.1 plugin in Unity 2019