stereolabs/zed-unity #65: Occlusion

GiuseppeLozioEllena, 2019-06-25:
Hi, I'm giuseppe Ellena.
i'm working on a project with Zed unity plugin. I have a problem with the occlusion algorithm. I take  two pictures of the result of the occlusion on a simple cube. It's possible to increment the precision of occlusion algorithm to avoid the effect that you can see on the picture2? I want to see correctly my hand over the cube like the picture1.  I think that the angle used in the occlussion algorithm for internal faces of shape is to big and this decrements​ the precision of the occlusion for the internal faces of shape.  What is a possible solution? 
thanks for help.
[image]
[image]

GiuseppeLozioEllena, 2019-08-29:
hi, 
I realize a opencv wrapper for unity to create a mask. 
the mask is the result of image subtraction of the same background and all frames capture from zed camera.  Now i work hard to add this mask to the unity shader, but there are so many things in connection with it that I hardly know where to begin.
[image]

Abrahamh08, 2019-09-04:
That's kind of a hacky solution imo. I wonder why the zed team hasn't looked into this after 2 months.

ckorris, 2019-09-04:
Pardon the lack of input here on GitHub; GiuseppeLozioEllena and I were chatting about this via email. 

Unfortunately, these kinds of occlusions are a natural challenge to stereo depth, primarily occurring when one camera can't see a part of a surface that the other can. Our FILL mode (turned on by default in Unity) attempts to guess the depth in such situations using the depth of surrounding pixels, and in many cases, this improves the image significantly (especially along the edges of arms, etc.) 

In principle, you can add more algorithms to detect these errors and fix them, as GiuseppeLozioEllena has. But it's a trade-off between accuracy and performance. We've added more filtering to detect the edges of hands, for example, and indeed the occlusion around hands is better than it used to be, but it's very difficult to cover all potential occurrences of this occlusion issue without impacting performance way too much. GiuseppeLozioEllena's solution works for their use case but the frame rate + latency would be detrimental to pass-through AR.

We're always looking for smart ways to improve the depth, but usually the best things you can do to alleviate this issue are:

- Try increasing the depth mode from PERFORMANCE to higher modes like QUALITY and ULTRA (in ZEDManager)
- Make sure both the subject and background are well-lit and well-textured 
- As much as you can, have the background's color/texture contrast with the subject's

closed 2022-04-27 by github-actions[bot]