stereolabs/zed-unity #70: [Feature Request] Allow custom tracking and camera setup using VR

jespersmith, 2019-07-10:
We're working on integrating the ZED Mini with a teleoperated mobile robot with mixed reality visualization of the state of the robot. An example of what we want to do can be found in this video published by MIT/CSAIL: https://youtu.be/4a-W3Od5-t8?t=16

We therefore need to disable the tracking functions inside ZEDMixedRealityPlugin.cs. We created our own implementation of ZEDMixedRealityPlugin.cs that provides the correct camera setup and tracking for our system by re-implementing the relevant functions called by ZEDManager.

Ideally, tracking (or specifically, ZEDMixedRealityPlugin) would be exposed as an interface where we could implement our own tracking and camera setup, so we don't get locked in in a specific version of zed-unity but can easily upgrade.

ckorris, 2019-07-10:
That's an interesting idea. 

Was it much work for you to get the full FoV of the VR headset while using the ZED's AR passthrough features? That's been a stumbling block for some people who aren't using straight pass-through, but in their case, they usually want the ZED image to still follow the person's head movements (unlike you). 

I ask because making ZEDMixedRealityPlugin inherit from an interface might give us a good opportunity to make another class like it that inherits from it, but without the FoV restriction like some people want. Kill two birds, etc.

jespersmith, 2019-07-11:
I was able to adjust the scaling of the final rendering quads which gave me a larger FoV by changing the ZEDReady function.

I agree that making ZEDMixedRealityPlugin inherit from an interface will provide the most flexible solution.

Blindsp0t-creative, 2020-01-17:
Hi jespersmith, 

Really interested in your post! Could you give more details on this "larger FoV" technique ?
+1 for the interface !

jespersmith, 2022-04-21:
We do not use the ZED MR features anymore, just render the quads in front of the user.

closed 2022-05-28 by github-actions[bot]