stereolabs/zed-unity #75: [Bug?] Depth Occlusion

stef-t, 2019-07-23:
For some reason I do not seem to be able to run the ZED in "depth occlussion off mode". I have set _Depth Occlusion_ to false (not ticked) in the Inspector on the `ZEDManager` component. 
From what I can tell the correct shader keywords are also set in each _Frame_ -- that is `"NO_DEPTH_OCC"` -- and the correct material is also linked in each _Frame_'s renderer. I also tried both _Forward_ and _Deferred_ shading with the exact same results.

The [docs](https://www.stereolabs.com/docs/unity/basic-concepts/) describe _Depth Occlusion_ as the following:
> Depth Occlusion: When enabled, virtual pixels can be covered up by real pixels, allowing a virtual cube to be behind your real table, for example. Turn off to make virtual objects always appear over the real world.

I specifically want virtual objects always appearing over the real world, but turning it off makes all virtual objects disappear and it looks like the opposite is happening -- the real world always appears on top of my virtual objects!

So I was wondering if anybody is experiencing the same or can help me pinpoint the issue I'm having.

For completeness: Turning depth occlusion on works as expected.

obraun-sl, 2019-07-24:
Hi,
Can you specify the OS (windows or linux) and the Graphics API used (OpenGL or DirectX) ?

stef-t, 2019-07-24:
Hello,
I'm using Unity on Windows with [OpenGLCore ](https://docs.unity3d.com/Manual/OpenGLCoreDetails.html).

stef-t, 2019-07-24:
Thank you for the hint! I didnt't think of the Graphics API. _Depth Occlusion_ is working as expected when used with DirectX11!

Seems like OpenGLCore was the culprit.

obraun-sl, 2019-07-24:
I will push the fix for OpenGL as it should work for this API as well

obraun-sl, 2019-07-24:
I pushed the fix in master (commit 27ccd58) . Can you check that if fixes openglcore ? (on my side, it did).

stef-t, 2019-07-24:
Your fix resolved the issue on my side as well. _Depth Occlusion_ is now working as expected, thank you.

closed 2019-07-24 by stef-t

KalydosENSIIE, 2021-05-12:
Hello,
I have the same issue.
I'm using unity 2019.4.26f1 with OpenGLCore on Windows 10.