stereolabs/zed-unity #77: GarbageMatte not rendering in Unity Build

horst-evans, 2019-08-16:
I have run into an issue that I just cannot seem to solve.
What I am trying to do is have a GarbageMatte (from the greenscreen functionality) appear in a scene at the start and act as a window into reality so that a user can view their keyboard while everything else around them is Virtual Reality. This is working exactly as I want it in the Unity Editor. However, once I build the project out, the GarbageMattes no longer work. Instead of projecting the capture from the ZED-Mini, the GarbageMatte in the build is just showing me the default gray material used in the edit mode. 

I am using a ZED-Mini camera in Unity 2018.4.6, Windows 10.

To get the GarbageMatte to load in the build, I use the script below to load the GarbageMatte once the scene opens and the ZED-Mini camera connects.

    // the gameobject that contains the camera and green screen manager
    public GameObject zedcamera;
    // Start is called before the first frame update
    void Start()
    {
        zedcamera.GetComponent<GreenScreenManager>().LoadGreenScreenData(true);
    }

Any guidance will be greatly appreciated.

SCREENSHOT OF EDITOR RUNTIME
[image]

SCREENSHOT OF BUILT OUT EXECUTABLE RUNNING
[image]

ckorris, 2019-08-19:
I found the issue, and we'll fix it in our next release. In the meantime, you can fix it by commenting out [line 345 of GreenScreenManager.cs](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/Examples/GreenScreen/Scripts/GreenScreenManager.cs#L345): 

```
#if !UNITY_EDITOR
        Debug.Log("Load Chroma keys");
        LoadGreenScreenData();
        UpdateShader();
        //CreateFileWatcher(""); <- This is the culprit, comment out this one. 
#endif
```

horst-evans, 2019-08-19:
This fixed my issue. My build is rendering correctly again!

closed 2019-08-19 by horst-evans