stereolabs/zed-unity #78: Not working on Windows Mixed Reality

GONBEEEproject, 2019-08-19:
Hi, because i'm japanese, sorry to terrible english.
Thanks for interesting device and SDK.
I'm enjoying them very much.

I've tried to use ZED mini with HTC Vive,Oculus Rift S and Windows Mixed Reality (Acer),
then Vive and Rift S have worked, but Windows MR haven't worked.

**Environment**
Windows 10 Home
Intel Core i7-6700
16GB DDR3 memory
NVIDIA GeForce GTX1070
SteamVR 1.6.10

Unity 2017.4.10f1
SteamVR SDK 1.2.3

**Tried**
I set up ZED mini based on [Getting Started](https://github.com/stereolabs/zed-unity#getting-started)
and replaced headset to Windows MR looking [Adding AR Passthrough](https://github.com/stereolabs/zed-unity#if-using-a-wmr-headset).

**Happened**
ZED mini looks activated and operate normally.
But Headset tracking doesn't work.
Screen can be generated normally.
but when move head, screen doesn't follow headset.

Maybe camera's(headset) rotation rotate twice, so screen moves anywhere.

sorry to japanese tweet.
this is capture movie of these movement.
When move head, the screen doesn't follow headset movement, 
and be black screen.

<blockquote class="twitter-tweet"><p lang="ja" dir="ltr">WinMRにスマホ突っ込んで撮影<br><br>ゴーグルにZED貼ってあって、<br>映像は回転してるのが見える<br>が、そのスクリーンがヘッドセットをトラッキングしてなくて動かない<br><br>謎〜〜〜 <a href="https://t.co/jCf5uAEIYA">pic.twitter.com/jCf5uAEIYA</a></p>&mdash; ごんびぃー@RefrainCore (@GONBEEE_project) <a href="https://twitter.com/GONBEEE_project/status/1159302636396408832?ref_src=twsrc%5Etfw">August 8, 2019</a></blockquote> <script async src="https://platform.twitter.com/widgets.js" charset="utf-8"></script>

Did anyone encounter this matter?
I couldn't solve this problem.
Please give me some ideas.

thanks,
GONBEEE

ckorris, 2019-08-19:
First, in order to use Windows Mixed Reality with the ZED Mini, you need to install SteamVR for Windows Mixed Reality: https://store.steampowered.com/app/719950/Windows_Mixed_Reality_for_SteamVR/

Once you've installed that, can you confirm that you can use SteamVR normally with your WMR headset, and all the tracking works? I'm wondering if that headset isn't configured correctly.

GONBEEEproject, 2019-08-27:
Thanks for reply

First, I changed Windows MR headset from Acer to Samsung Odyssey, then head tracking had fixed
and worked normally!!

I tried re-installing Window MR application and WMR for SteamVR, but head tracking didn't recover.

but when i tried my new Odyssey, head tracking worked normally without re-installing applications.
I think Acer WindowsMR is one of the oldest WMR headset, so maybe tracking system(or hardware) is different from recent WMR headsets.

thanks for your cooperation

closed 2019-08-27 by GONBEEEproject