stereolabs/zed-unity #79: [Q] Delaying Image Stream

stef-t, 2019-09-09:
Hello,

I'm currently working on combining a tracking system with the ZED's visualization capabilities. One part of this is the synchronisation between the ZED's imagestream and the input. 
As it turns out there are a few milliseconds delay between the time when I get updated tracking system data and the time when the ZED updates its imagestream. This basically results in the virtual objects, which are attached to tracked targets of this tracking system, slightly lagging behind its real counterparts in the imagestream of the ZED. 

I was wondering if there exists a method or scripting component, which allows to delay the time between the recording of new images with the ZED and those new images actually being displayed. I'm aware that a too high delay of this sort results in a _bad_ experience aka motion sickness, etc., but I want to give this a try to see if these virtual representation of the real tracked targets can be perfectly aligned in the the final ZED imagestream.

Kind regards.

ckorris, 2019-09-09:
We don't have an existing script in place to do this (in fact our timewarp system does basically the opposite) but it should be possible. Basically you would be storing the ZED images in a buffer instead of applying them directly to the final texture, and then actually applying them once your tracking system has updated. 

Normally, we create the ZED textures from [ZEDCamera.CreateTextureImageType()](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/NativeInterface/ZEDCamera.cs#L1329) and then apply that to the Mat_ZED_Forward_Lighting material on the canvases in your ZED rig prefab. 

What's important to note is that once you have that texture, we update it behind-the-scenes in our .dll every time there's a new image available (it will update right away even if you do nothing in Unity). So you just call it once at the start, and you would call [Graphics.Blit](https://docs.unity3d.com/ScriptReference/Graphics.Blit.html) to copy from that texture to another when you want to buffer it. 

So in short, I'd make a class that's subscribed to the [ZEDManager.OnGrab](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs#L921) event, and when you do, blit that texture you made at start into a separate buffer RenderTexture image you created the regular way in Unity. Then when your tracking update is done, you can Blit that buffer image to _another_ RenderTexture that would be set to Mat_ZED_Forward_Lighting. 

If you want to get more advanced with the timing, we have some [timestamp functions in ZEDCamera](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/NativeInterface/ZEDCamera.cs#L1178-#L1200) that you can use to get when the image was taken with some more precision. I'm not sure you'll need that but just so you know it's there.

stef-t, 2019-09-10:
Thank you for this insightful answer! 

The timewarp system was actually the reason I thought there may be a system which does the exact opposite. Thank you for outlining the steps needed to implement this buffer, I will go ahead and try to implement that for testing.

I will leave this issue open for a short while, in case I have some follow up questions.

ckorris, 2019-09-10:
Sounds good!
I also just realized that you'll probably want to do the same with the depth images, assuming you're using occlusion. In that case, you'll need to get the depth image with [CreateTextureMeasureType() ](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/NativeInterface/ZEDCamera.cs#L1390) instead of CreateTextureImageType(). That's important because there _is_ a depth map image available from CreateTextureImageType, but it's only meant for previewing/debugging the depth (like you see in the Depth Viewer app), is a lot less accurate, and the shader isn't written for that type of data.

ReallyRad, 2020-03-03:
@ckorris @stef-t  I'm trying to implement something similar, but with a delay of several seconds. I'm wondering if using RAM instead of graphics memory could possibly drive the framerate down?

thanks

closed 2021-09-01 by stef-t