stereolabs/zed-unity #83: Vuforia + Zed Mini?

Naphier, 2019-09-25:
We're looking to implement Vuforia with the Zed Mini. We have a license that allows Vuforia to support "external cameras" which, in theory, should mean we can use the Zed Mini. Has anyone accomplished this?

ckorris, 2019-10-11:
I'm afraid we haven't tested the external camera support in Vuforia internally - only the base Unity plugin, and long enough to know that it's not possible to support without something like their external camera feature. 

That said, we just released [v2.8.2](https://github.com/stereolabs/zed-unity/releases/tag/2.8.2) of the ZED Unity plugin, which includes ArUco marker detection via OpenCV if you also import the [OpenCV for Unity](https://assetstore.unity.com/packages/tools/integration/opencv-for-unity-21088) package. 

I assume you're fairly committed to Vuforia if you've invested in that license, but perhaps this could be a workaround, at least temporarily?

Naphier, 2019-10-11:
Thanks, Chris.
Yes, Vuforia has some features that we're committed to not writing with OpenCV. Slightly more complex than just ArUco detection.
We do have the external camera license so we'll need to build a special "driver" to work with the Zed. Still hoping someone out there has worked on this so that we can share experiences. 
Technically it seems possible, but re-inventing the wheel is what we're trying to avoid. 
Unfortunately, PTC/Vuforia isn't much help in this regard :(
Thanks!

478161648, 2019-11-06:
I have not purchased OpenCV For Unity yet.
can you tell me the ArUco can be used with Oculus Rift or HTC VIVE?
Thank you.

ckorris, 2019-11-11:
Yes, it can be used with any headset that's compatible with the ZED Mini for pass-through AR. Though note that because OpenCV is performance-intensive, you may drop to a lower FPS than is comfortable for VR or AR, depending on your system specs. 

Also note that you can use the free trial version of OpenCV for Unity with our plugin, and the only limitation is that you can't build it into a final app (only run in the editor). You can download the trial version here: https://enoxsoftware.com/opencvforunity/get_asset/

JurjenBraam, 2020-05-11:
Does anyone already got Vuforia working with the ZED Mini?

closed 2022-04-27 by github-actions[bot]

AndreAhmed, 2025-02-01:
Anyone succeeded to get Vuforia with ZED Mini?
@JurjenBraam 
@ckorris 
@Naphier

Naphier, 2025-02-01:
I honestly don't recall. It was a small experiment and many years ago now.

JurjenBraam, 2025-02-04:
Same for me. Eventually I switched to ARKit for more reliable tracking