stereolabs/zed-unity #84: Retriev 3 channels image

rmtttt, 2019-10-10:
Hello,

Is it possible to retriev from sdk a 3 channels Mat (MAT_TYPE_8U_C3) instead of 4 channels (MAT_TYPE_8U_C4) like this?

```
Mat leftImage;
while (true) {
        if (zed.grab() == SUCCESS) {
                zed.retrieveImage(leftImage, VIEW_LEFT);}}
```

ckorris, 2019-10-11:
For VIEW_LEFT, I'm afraid it's not - the number of channels of an image is fixed to that image type, and it's 4 channels for that type. 

You'd have to convert it afterwards, or change how the image is sampled. May I ask why you need the three-channel version?

rmtttt, 2019-10-11:
I'm working on application which runs an object detector on ZED image, and result of detector I use for pointcloud analysis. Model of detector supports RGB image only. Since all together it's quite heavy, I want to avoid image converting.

Ideally I need to get IntPtr on GPU for RGB texture.

How to get IntPtr on GPU is clear for me, but is it possible to request 3 channels ZEDMat?

ckorris, 2019-10-11:
I'm afraid this is not possible, for the reasons I mentioned in my last post. Sorry about that.

rmtttt, 2019-10-14:
Okey, thanx.
And may I ask what kind of data you store in 4th channel?

ckorris, 2019-10-15:
The fourth channel is the alpha channel. It should pretty much always be 1, so it's safe to remove that channel if you put it into an RGB-formatted texture.

closed 2022-04-27 by github-actions[bot]