stereolabs/zed-unity #86: ZED visible range question

KawaStudy, 2019-10-18:
Hi, I'm currently working on with Unity 2017.2 and Oculus Rift.
I followed the "Adding AR Passthrough" section in README.md and tried a Planetarium scene.
They are so cool to see but visible range seems a little bit narrow.
Is there any way to get more wider area to see? Or is this the limit that I can get?

I attached the image how I see from Oculus HMD.
[image]

ckorris, 2019-10-18:
The reason you see the borders is because the FoV of the ZED Mini is less than the FoV of your headset. 

We experimented with this a lot when we first created the pass-through AR. We did try stretching out the image to the edges, tried warping it in different ways, tried blurring the edges by interpolating them from the visible area, etc. But ultimately we found that it just wasn't a very comfortable experience unless you displayed it the way we did, where every visible pixel matches up realistically with the real-world pixel behind it. 

If you _did_ want to experiment with making the image bigger, you can do that in ZEDMixedRealityPlugin.cs. In `ZEDReady()`, there's a call to `ComputeSizePlaneWithGamma()` ([line 422](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/MR/ZEDMixedRealityPlugin.cs#L422) as of writing) which calculates how big the final viewing planes should be. A simple way to change the size would be to add a line afterward that multiplies that `scaleFromZED` Vector2 by 1.2 or 1.5 or something. The subsequent calls to `scale()` are where the sizes actually get changed.

KawaStudy, 2019-10-19:
I tried the way to make the image bigger you told me, and likely understand why you told me the stretching a image method would result in uncomfortable experience.

> The reason you see the borders is because the FoV of the ZED Mini is less than the FoV of your headset.
This gives me a lot of comprehension and I'd appreciate it.

I'll go with a default setup around here.
Thanks for you answer!

closed 2019-10-19 by KawaStudy