stereolabs/zed-unity #87: Occlusion without displaying the real world

flaming910, 2019-10-18:
I am developing something in which I want to use the zed but I don't want to output the actual image from it. I want to be able to have depth occlusion without having the unity camera display what is captured from the real world with the camera. Is it possible to do this?

ckorris, 2019-10-18:
You can avoid showing the RGB camera input by commenting out the line in ZEDRenderingPlane where we assign the `textureEye` material to the `_MainTex` property of the main shader (ZED_Forward_Lighting usually). This is on [line 655](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/Display/ZEDRenderingPlane.cs#L655) as of writing: 

`matRGB.SetTexture("_MainTex", textureEye);`

With that commented out, no image gets assigned to that material's color. However, instead you just see a default gray color. Here's me waving in front of a virtual cube after making that change:  

[image]

I'm assuming this isn't quite what you want. Could you describe what you'd like to see instead when an object is occluded?

flaming910, 2019-10-18:
I want it to just be black when its occluded. So instead it would like this

[image]

ckorris, 2019-10-18:
Okay, then you can change the line I said to comment out to assign a black texture to it instead. 

Or even easier, in [ZED_Forward_Lighting.shader](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/Lighting/ZED_Forward_Lighting.shader), you can comment out assignment to `outColor` near the end ([line 144](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/Lighting/ZED_Forward_Lighting.shader) currently) and replace it with a black color, like so: 

```
//outColor.rgb = c;
outColor.rgb = float3(0, 0, 0);
```

flaming910, 2019-10-18:
That is great. Would it be possible to output only the black part to lets say a render texture? So anything that isn't occlusion is transparent?

ckorris, 2019-10-18:
It sounds like the best way to do what you're trying to do is with a depth mask. Basically like this shader: https://wiki.unity3d.com/index.php/DepthMask

That shader is set to render before the other objects in question, and writes to the depth buffer, but does _not_ render to the color buffer. So it's invisible, but objects rendered afterward just won't render to the color buffer where they're occluded. 

Doing the following should get the effect you want, or pretty close: 
- Remove all color output in the [ZED_Forward_Lighting](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/Lighting/ZED_Forward_Lighting.shader) fragment shader (`out fixed4 outColor : SV_Target` in the declaration and anything that references it) 
- Add `ColorMask 0` up top, near where `ZWrite On` is
- Lower the RenderQueue value to be lower than the materials on all the rendered virtual objects (that can be occluded). You can either put something like "Geometry-10" in the shader itself, or in the Mat_ZED_Forward_Rendering material, override it to something like 1990 (which is the same as Geometry-10, since Geometry = 2000)

That should have the effect of virtual objects only appearing if it's closer than the real world as seen from the ZED, but without you actually seeing the ZED. If not, though, let me know, as this sort of thing can be finnicky. 

If you need to put this into a mask of some sort, just set the output texture of the ZED camera to a RenderTexture and go from there.

flaming910, 2019-10-21:
So I added `ColorMask 0` into SubShader, and also changed tags to be `Tags{ "LightMode" = "Always" "Queue" = "Geometry - 10"}`. 

I also changed the function decleration of the fragment shader to `void frag_surf(v2f_surf IN, out float outDepth : SV_Depth)` and commented out these two lines:

- `UNITY_INITIALIZE_OUTPUT(fixed4,outColor);`
- `outColor.rgb = c;`

When I run the camera loads up and everythings black and slowly shifts to a dark grey. There are multiple virtual objects in view at different distances and none of them are showing up.

ckorris, 2019-10-21:
I copied those same changes on my end, and I got what I believe to be the intended result, below. I changed the clearing color of the cameras to blue to show that the black was from the lack of anything being rendered, not from improperly writing a black color to the color channel: 

[image]

So I think there's something else getting in the way of doing what you want in your scene. Perhaps could you give a screenshot of your scene and Hierarchy? 

The shifting to a dark gray is quite unexpected. That can happen if a RenderTexture is being updated with empty data, and a color conversion causes the color to slowly change. 

Also, are you using pass-through AR? I'm wondering if something is happening with the interaction between the ZED_Rig_Stereo and the hidden AR camera rig (that you can enable and see by checking "Show AR Rig" in ZEDManager's Advanced Settings). If so, I'd try turning off VR support in the project and seeing if you still have this issue. 

If not, do you have the Target Texture values of the cameras in the rig set to something? If so, I'd try clearing it and seeing if you get the same results. 

FYI, if using pass-through AR, the Target Texture values are automatically set to RenderTextures that we create at runtime from the hidden AR rig - so not only would they not be empty, but they'd clear any other RenderTexture target that you set up there. 

Side note: I did also try displaying the skybox in the background instead, but it was covered up because _every_ pixel in view does get a depth value assigned, even if it's really far away, which means the depth isn't infinite and therefore the skybox isn't shown. In the latest version of the plugin, you could solve this by playing with the Max Depth slider in the material, which will cull any pixels beyond the given range (in meters).

flaming910, 2019-10-22:
So I got it working in another scene, but bringing it back into my original scene ended up returning the weird grey background issue. I used the frame debugger and found out that for some reason the post processing shader causes that to happen, unsure why. 

How would I put this into a depth mask kind of thing now? because putting it into a render texture still keeps the virtual objects that are visible in it, and I am unsure of how I would go about removing them for the depth mask.

ckorris, 2019-10-22:
My thoughts were that you could avoid the rendertexture and second camera and just do this on your main camera, since otherwise you would have twice as many draw calls for each virtual object (once in the ZED camera just to test its depth, and again in the main camera) whether or not it's occluded. That would result in a performance hit. 

But if that's not possible, I can think of a few ideas. But they would all pretty much involve using an image effect shader on the final result to create the mask. This is because if you want to get the depth of the virtual objects for occlusion, you have to draw them, which also gives you their color pixels, unless you plan on replacing all the shaders in those objects (which is usually too inconvenient). 

So basically in an OnPostRender call on a script attached to that camera, you could call Graphics.Blit from the RenderTexture to a new texture, using a material with such an image effect shader. 

Some ways the shader could work: 

1. Make the background color something bright and distinct, like RGB 0,0,255, then have the image effect shader convert all green values to empty values, and everything else to whatever mask value you want
2. In ZED_Forward_Lighting, remove the `ColorMask 0` line, and add `ZTest Greater` so then it renders only if the pixel is _behind_ another pixel, not in front, set the render queue to something greater than the rest (like Geometry+10). Then have the output color be something distinct like that green color, and use the image effect shader to filter out everything else

Then the output of that blit would be the mask to use.

flaming910, 2019-10-23:
I actually realized with the new ZED_Forward_Lighting shader I might not need to make a separate mask shader, but if its needed thank you for explaining how I'd go about doing it.

ckorris, 2019-10-23:
Ah, I could've been more clear about that, sorry.

flaming910, 2019-10-23:
Oh no its fine it didn't register in my head that the new shader does exactly what I want.

flaming910, 2019-10-25:
Is it possible to offset the occlusion? The camera is going to be placed above the user's eyes(essentially forehead level) so the occlusion doesn't really line up with their hands(going for a mixed reality thing)

ckorris, 2019-10-25:
It is. You can move the Frame objects in the ZED rig to be at the position you need. 

However, ZEDRenderingPlane (the script attached to each camera) does position the frame on its own once the ZED loads, to make sure they're at the optical center of the camera, starting [line 501](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/Display/ZEDRenderingPlane.cs#L501-#L507) as of writing: 

```
float plane_distance =0.15f;
Vector4 opticalCenters = zedCamera.ComputeOpticalCenterOffsets(plane_distance);

if (side==0)
    canvas.transform.localPosition = new Vector3(opticalCenters.x, -1.0f * opticalCenters.y,plane_distance);
else if (side==1)
    canvas.transform.localPosition = new Vector3(opticalCenters.z, -1.0f * opticalCenters.w,plane_distance);	
```

So you could position the frames how you want them before runtime and then just comment those lines out. Or if you want, and it's probably a better idea to make sure you account for camera calibration more,  you could add a Vector3 field to the script, and add that value to the localPosition of each canvas after the above code runs. Something like:

`canvas.transform.localPosition = canvas.transform.localPosition + customOffsetVector;`

flaming910, 2019-10-25:
I tried to just change the position manually afterwards, but then the rendering plane isn't filling the whole view so I see behind the plane a little bit at the edge of the screen(unless changing it before hand will resize it accordingly?)

flaming910, 2019-10-28:
I've gotten it so its almost perfect, and in my attempts to improve it I tried changing the plane to use the right eye, but then with my new shader I just get a black screen for some odd reason, any idea why that could be happening?

ckorris, 2019-10-28:
Can you tell me what you changed?

flaming910, 2019-10-28:
I've just modified the scale and also added a slight offset at the end of ZEDReady. Haven't touched the shader, just went to the Camera_Left and under rendering plane set it to RIGHT instead of LEFT

ckorris, 2019-10-29:
Ah, if you're using the ZED_Rig_Stereo rig, it might be because changing the eye setting in ZEDRenderingPlane is preventing ZEDManager from detecting that the GameObject is indeed a stereo rig intended for AR, thus stopping it from setting up a few things to make pass-through work. That happens in ZEDManager.CheckStereoMode() with the relevant parts being in [line 1071](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs#L1071) as of writing: 

```
        if (cameraLeft == null || cameraRight == null)
        {
            foreach(Camera cam in monocams)
            {
                ZEDRenderingPlane rendplane = cam.gameObject.GetComponent<ZEDRenderingPlane>();
                if (!rendplane) continue;

                if(!cameraLeft && (rendplane.viewSide == ZEDRenderingPlane.ZED_CAMERA_SIDE.LEFT || rendplane.viewSide == ZEDRenderingPlane.ZED_CAMERA_SIDE.LEFT_FORCE))
                {
                    cameraLeft = cam;
                    camLeftTransform = cam.transform;
                }
                else if(!cameraRight && (rendplane.viewSide == ZEDRenderingPlane.ZED_CAMERA_SIDE.RIGHT || rendplane.viewSide == ZEDRenderingPlane.ZED_CAMERA_SIDE.RIGHT_FORCE))
                {
                    cameraRight = cam;
                    camRightTransform = cam.transform;
                }
            }
        }
```

Basically, one of the steps to determining that you're using a stereo rig is to make sure that you have one ZEDRenderingPlane set to a Left eye and one ZEDRenderingPlane set to a Right eye. If that condition is not true, it won't create a second, normally-hidden AR rig needed for pass-through. So you could remove those conditionals from ZEDManager or just use the left side for the left eye and right side for the right eye. 

If you're not actually using ZED_Rig_Stereo, let me know.

flaming910, 2019-10-29:
I'm currently using the mono rig(might be swapping to stereo rig later on)

ckorris, 2019-10-30:
I tried to replicate this but I can't. Can you give some more information about your setup?

If you just make an empty scene with ZED_Rig_Mono and change the ZEDRenderingPlane's View Side value to Right, and hit play, can you see anything?

flaming910, 2019-10-31:
When I set to Right it displays if I use the original forward lighting shader. If I use the modified one for the occlusion mask it then just displays black. If I turn off occlusion while using that shader then virtual objects show up again but occlusion is no longer functioning.

flaming910, 2019-10-31:
Also, slightly off topic, but is there a way to use zedCamera.RetrieveImage and get an image that is RGBA instead of BGRA?

ckorris, 2019-10-31:
That's strange, I'm looking through the code that changes when you set the view to left/right and I don't see anything that should be breaking it - it might be a deeper issue. I'll test this afternoon. Just to make sure I can test it exactly, could you upload your version of the shader? (If you don't reply in time I'll remake it, but getting the exact same shader would be ideal)

Regarding zedCamera.RetrieveImage as an RGBA, I'm afraid you can't do this directly as that's a direct call to our C++ SDK, which outputs textures in BGRA format. But once you have it as a Texture object (usually from CreateTextureImageType but also if you use RetrieveImage and apply it to a Texture) then you can quickly convert it to RGBA using an image effect shader: 

1. Make an Image Effect shader. The default shader Unity makes when you create one in the Assets folder is pretty close to what you need. You just need to change the col value in the frag shader to "swizzle" from BGR to RGB. Like this: 

```
fixed4 frag (v2f i) : SV_Target
{
    fixed4 col = tex2D(_MainTex, i.uv);
    col.rgb = col.bgr;
    return col;
}
```
2. Make a material for that shader
3. In a script, add a reference to that material (either through a field you set in the Inspector, a Resources.Load call, etc.) 
4. Call Graphics.Blit() with three arguments: the original BGRA ZED texture, a new target RenderTexture, and the material you just made. (Sometimes you can set the destination texture as the same as the target texture, but whether that will work depends on a few things, so it's cleaner to make a new RenderTexture with the same settings)

flaming910, 2019-10-31:
Here is the shader:

`//======= Copyright (c) Stereolabs Corporation, All rights reserved. ===============
 // Computes lighting and shadows and apply them to the real
	Shader "ZED/ZED Forward Lighting Occluded"
{

	Properties{
		[MaterialToggle] directionalLightEffect("Directional light affects image", Int) = 0
		_MaxDepth("Max Depth Range", Range(1,20)) = 20
	}
		SubShader{

		ZWrite On
        ColorMask 0

		Pass{
		Name "FORWARD"
		Tags{"LightMode" = "Always"}

		Cull Off
		CGPROGRAM
		// compile directives
#pragma target 4.0

#pragma vertex vert_surf
#pragma fragment frag_surf

#pragma multi_compile_fwdbase
#pragma multi_compile_fwdadd_fullshadows

#pragma multi_compile __ NO_DEPTH_OCC

#include "HLSLSupport.cginc"
#include "UnityShaderVariables.cginc"

#define UNITY_PASS_FORWARDBASE
#include "UnityCG.cginc"

#include "AutoLight.cginc"
#include "../ZED_Utils.cginc"
#define ZED_SPOT_LIGHT_DECLARATION
#define ZED_POINT_LIGHT_DECLARATION
#include "ZED_Lighting.cginc"

		sampler2D _MainTex;

	struct Input {
		float2 uv_MainTex;
	};

	float4x4 _CameraMatrix;
	sampler2D _DirectionalShadowMap;

	struct v2f_surf {
		float4 pos : SV_POSITION;
		float4 pack0 : TEXCOORD0;
		SHADOW_COORDS(4)
			ZED_WORLD_DIR(1)

	};

		float4 _MainTex_ST;
		sampler2D _DepthXYZTex;
		float4 _DepthXYZTex_ST;
		sampler2D _MaskTex;
		int _HasShadows;
		float4 ZED_directionalLight[2];
		int directionalLightEffect;
		float _ZEDFactorAffectReal;
		float _MaxDepth;

		// vertex shader
		v2f_surf vert_surf(appdata_full v) {

		v2f_surf o;
		UNITY_INITIALIZE_OUTPUT(v2f_surf,o);
		o.pos = UnityObjectToClipPos(v.vertex);

		ZED_TRANSFER_WORLD_DIR(o)

			o.pack0.xy = TRANSFORM_TEX(v.texcoord, _MainTex);
		o.pack0.zw = TRANSFORM_TEX(v.texcoord, _DepthXYZTex);

		o.pack0.y = 1 - o.pack0.y;
		o.pack0.w = 1 - o.pack0.w;

		TRANSFER_SHADOW(o);

		return o;
	}
	// fragment shader
	void frag_surf(v2f_surf IN, out fixed4 outColor : SV_Target, out float outDepth : SV_Depth) {
	    UNITY_INITIALIZE_OUTPUT(fixed4,outColor);
		float4 uv = IN.pack0;

		float3 zed_xyz = tex2D(_DepthXYZTex, uv.zw).xxx;

		//Filter out depth values beyond the max value. 
		if (_MaxDepth < 20.0) //Avoid clipping out FAR values when not using feature. 
		{
			if (zed_xyz.z > _MaxDepth) discard;
		}

	#ifdef NO_DEPTH_OCC
	    outDepth = 0;
	#else
		outDepth = computeDepthXYZ(zed_xyz.z);
	#endif

		fixed4 c = 0;
		float4 color = tex2D(_MainTex, uv.xy).bgra;
		float3 normals = tex2D(_NormalsTex, uv.zw).rgb;

		//Apply directional light
		color *=  _ZEDFactorAffectReal;

		//Compute world space
		float3 worldspace;
		GET_XYZ(IN, zed_xyz.x, worldspace)

		//Apply shadows
		// e(1) == 2.71828182846
		if (_HasShadows == 1) {
			//Depends on the ambient lighting
			float atten = saturate(tex2D(_DirectionalShadowMap, fixed2(uv.z, 1 - uv.w)) + log(1 + 1.72*length(UNITY_LIGHTMODEL_AMBIENT.rgb)/4.0));

			c = half4(color*atten);
		}
		else {
			c = half4(color);
		}

		//Add light
	    c += saturate(computeLighting(color.rgb, normals, worldspace, 1));

		c.a = 0;
        // outColor.rgb = c;
		outColor.rgb = float3(0, 0, 0);

	}

	ENDCG

	}
	}
		Fallback Off
}
`

And I want to convert it to RGB for image processing reasons, and I have some things I'm going to test to see how well they perform.

flaming910, 2019-11-01:
Actually found out I do not need to convert to RGB and that what I'm doing does not care for color and is only interested in color intensity

ckorris, 2019-11-01:
Figured it out: There's an optimization in ZEDManager that disables depth measurements from the right side if you're not using AR pass-through mode, since otherwise you'd usually just use the left side. So if you only have a camera using the right side and/or VR is disabled, it won't get launched. 

You can fix that by adding a default value to InitParameters.enableRightSideMeasure in ZEDCommon ([line 977](https://github.com/stereolabs/zed-unity/blob/50f42e09a6f4263f88ce949009a713f765a4abae/ZEDCamera/Assets/ZED/SDK/NativeInterface/ZEDCommon.cs#L977) as of writing) to be true. 

Note that if you go back to using the left side, you'll want to change this back as it costs performance. (For that reason it would be better to use the left anyway if using the right side doesn't make the occlusion more accurate)

I'll change this to be smarter about setting that parameter in a future patch.

flaming910, 2019-11-01:
Ah ok, how much of a performance hit is using a stereo rig over using the mono rig. Also, what are some things I can do to improve performance that wouldn't effect tracking too much

ckorris, 2019-11-01:
I don't have too exact of a figure for how big of a hit it is. Maybe @obraun-sl can say? 

Otherwise, regarding performance, the plugin's default settings are all optimized for performance, so there's not a whole lot to do as far as the ZED/ZED plugin is concerned that would make a big difference, other than things you probably don't want to change (lower the requested FPS, disable tracking/spatial memory, etc.) If you're having performance issues, it may be best to look at your own settings/scripts/shaders and go through Unity best practices. Example: https://docs.unity3d.com/Manual/OptimizingGraphicsPerformance.html

obraun-sl, 2019-11-04:
Compared to the mono rig, the stereo rig will basically add two things that could decrease a bit performances : 
- Right depth is computed to make the occlusions on the right view (in mono, there is no right view). This is done in the zed sdk by enabling the right side measure ( `InitParameters.enableRightSideMeasure = true` ) . It' not as big as computing the left side depth since some of the work is re-used for right side. I would say that it increases about 20% of the total time at maximum.
- On Unity, the stereo rig will have 2 camera rendering jobs instead of one for mono rig. This is already well optimized in Unity (and Unity is made to have multiple camera rendering anyway), therefore it does not impact performances too much. 

To sum up and regarding the tests we have made, I would say that globally using the stereo rig will add 20-25% of the mono rig time to the pipeline. For example, if the whole pipeline takes 10ms in mono rig, the stereo rig will reach around 12.5ms at max.

flaming910, 2019-11-05:
I was wondering if you could explain the different settings under spatial mapping and advanced settings and their performance impacts if any?(Mainly the ones that have some level of performance impact)

flaming910, 2019-11-11:
@ckorris So I'm messing about with the ZEDArUcoDetectiongManager that works with OpenCV, and was wondering how I could introduce an offset to be able to do the same thing I did with the rendering plane and occlusion.

ckorris, 2019-11-11:
You can find more about the spatial mapping module in the non-Unity section of the docs: https://www.stereolabs.com/docs/spatial-mapping/

Regarding the offset, you should be able to add the same offset position and rotation to the pose of the object when it's given to the [MarkerObject](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/Examples/OpenCV%20ArUco%20Detection/Scripts/Core/MarkerObject.cs) class (or a class that inherits from it. 

For example, MarkerObject_MoveToMarkerSimple has this method that's called when it's marker is detected: 

```
public override void MarkerDetectedSingle(Vector3 worldposition, Quaternion worldrotation)
    {
        gameObject.SetActive(true); 

        transform.position = worldposition;
        transform.rotation = worldrotation;
    }
```

You could just multiply worldrotation by the offset rotation. 

If you also have a positional offset, you'll want to first get that in world space (so multiply it by the rotation of the ZED camera) and then you can just add it to worldposition. 

If I understand what you're looking to do, that should work.

flaming910, 2019-11-11:
So in my current setup, the overlap between the camera view and the user's MR view is shown in this image(not perfectly accurate but its just an idea):
[image]

So currently even if I offset the position or rotation, things that are out of view would still show up in the user's view. I want to be able to scale/offset things in a way that would cause only things that are within the users's view(with slight room for error) to show up in their view, other things can still be tracked but I don't want the user to see them.

ckorris, 2019-11-11:
I'm afraid I don't understand what you mean. What does the red box indicate in the image?

flaming910, 2019-11-11:
So I have a MR headset, where there is only partial overlap between what the screen would display to the user and what the zed camera sees. That red box is that partial overlap

ckorris, 2019-11-11:
So if I understand, if you did what you intended, you would only see some sort of virtual world in all the areas outside of the box. But within that box, you would see the real world, and you would only want to see that portion within the red box - the bottom of the wine glass, part of the can, etc. Whereas right now you're seeing the entire ZED image shrunk into that box. Is that correct?

flaming910, 2019-11-11:
Uh not exactly. So in the red box I want it to be the virtual world overlapping with the real world in an accurate fashion. So it would only show me that portion of the real world within the box along with whatever is tracked. The rest of it doesn't matter.

Currently its all shrunk into the box yes

ckorris, 2019-11-11:
Ah okay. It sounds like you'll need to get rid of the offset in order to avoid that scaling. 

What you could do is: 

1. Compute a mask texture of where exactly you want the ZED to be visible. Then sample that in the ZED_Forward_Lighting shader, and discard the pixel if it's not part of the mask. 

2. Use the [stencil buffer](https://docs.unity3d.com/Manual/SL-Stencil.html). Have a separate plane object that represents that "mixed" area, and give it a material with a shader that would draw before the ZED image (normally its RenderQueue needs to be less than 2000, like "Geometry-1"). That shader would always write a specific value to the buffer (Ref X, Comp Always, Pass Replace , where X is whatever arbitrary number you want above zero). Then you'd add the Stencil section to the ZED_Forward_Lighting shader, where it needs that value to be that same number (Ref X, Comp Equal)

closed 2020-01-23 by flaming910