stereolabs/zed-unity #88: When switching git branches you cannot "unlink" sl_unitywrapper.dll while Unity Editor is running

Naphier, 2019-10-23:
This causes issues with our continuous integration build system. No other DLL locks like this when Unity Editor is not actually running. I'm not sue what process is locking up your dll. But I assume that it's something like those config windows that pop up when a dev doesn't have the SDK installed on their machine. Please consider this sort of thing for future releases.

ckorris, 2019-11-01:
I'll definitely be taking a look at this - in addition to the inconvenience it's caused you, I suspect it's related to a crash we occasionally get after the editor has been open for awhile and the ZED has been opened many times. 

In the meantime, a quick hunch: can you try setting `ZEDManager.wrapperVerbose` to `false`? ([line 62](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs#L62) as of writing)

That line causes the .dll to open a log file that doesn't get finalized until Unity is closed.

Naphier, 2019-11-04:
Still won't unlink.

closed 2022-04-27 by github-actions[bot]