stereolabs/zed-unity #89: Ubuntu16.04 ZED SDK is not installed or needs to be updated

wsakobe, 2019-10-26:
My computer has installed ZED SDK 2.8.3 and it works well
When I run an example of zed unity plugin 2.8.2 it comes out this issue, please tell me how to solve this problem(sad).

wsakobe, 2019-10-26:
Unity 2019.1.0f2
ZED SDK 2.8.3
zed unity plugin 2.8.2
TITAN V
CUDA 10
NVidia Drivers 410

obraun-sl, 2019-10-28:
Hi ,
Did you use the linux_compatibility branch ?

wsakobe, 2019-10-28:
Thanks for your answering!
I saw that branch now.And i'd like to confirm that does it means I must program my project in Windows and can only test in Ubuntu?

obraun-sl, 2019-10-28:
With that branch, you can develop your app on windows or linux and build for windows and/or linux
both plugin library (sl_unitywrapper.dll/.so) are in the branch.

wsakobe, 2019-10-29:
Oh that's cool, let me have a try. Thanks a lot!

closed 2022-04-27 by github-actions[bot]