stereolabs/zed-unity #90: There is no texture display on the frame.

478161648, 2019-11-06:
I update the zed sdk to 2.8.3 and import the ZED Plugin for Unity v2.8.2. I can't see the real world. the frame plane is white... what should i do with the case?
[image]

obraun-sl, 2019-11-06:
Hi,
Remove the mainCamera from the project. The ZED_Rig_Mono already contains the main camera.

478161648, 2019-11-06:
> Hi,
> Remove the mainCamera from the project. The ZED_Rig_Mono already contains the main camera.

Thank you, the mainCamer is not the key point.I run the sample,the frame was still blank.
I removed CUDA9.0 and installed CUDA10.0 and the SDK for CUDA10.0, and the problem be solved.
maybe the CUDA9.0 and the SDK for CUDA9.0 have something wrong.

closed 2019-11-06 by 478161648