stereolabs/zed-unity #92: Feature Request: Retrieve Image - BGR

ed-spatialsauce, 2019-11-13:
Hey,

If possible it would be great to be able to get a BGR mat with RetrieveImage. Currently I get a BGRA mat, but I then after I copy the byte[] i need to convert it from BGRA to BGR.

Thanks

ckorris, 2019-11-13:
Our SDK only provides a BGRA version because getting this data from the GPU is much more efficient than a BGR texture for memory address reasons - a BGR color is 24 bits instead of 32, so the GPU can end up looking at two contiguous addresses for the same color value. 

So if we provided a BGR mat, it would be doing pretty much the same conversion that you would be applying, and wouldn't be much better (if at all) for performance. Though perhaps a conversion helper method could make this more convenient to do? 

May I ask why you need the BGR version?

ed-spatialsauce, 2019-11-13:
Hey Chris,

I thought that might be the case. I'm passing the BGR into Wrnch.

When I have time I'll try write a speedy converter, or I could ask the Wrnch team to add something on their side.

ckorris, 2019-11-14:
Ah, you're the second person this month to ask for this specifically for handing off to an AI library. I'm not sure if the other user was using Wrnch but could be. 

That'd be convenient if their end accepted BGRA, but their situation might be similar, where there's a good reason they don't want four channels, and the best solution is still just to convert it as fast as you can. 

If you write a speedy converter in Unity, would you mind sharing it here?

ed-spatialsauce, 2019-11-14:
Sure, I'll reply back when I write something.

closed 2022-04-27 by github-actions[bot]