stereolabs/zed-unity #94: CreateTextureImageType

dimll, 2019-12-04:
Hello,

My problem is similar with the issue #67 but the solution that is mentioned isn't fitting my project. I would like to use the Texture2D that is given from zed and map it into a mesh. The texture generated from `CreateTextureImageType(sl.VIEW.LEFT)` is blueish and it seems that the R and B channels need to be reversed. Changing the texture format on the ZEDCamera.cs script as shown below isn't working. 

`//m_Texture = new Texture2D(width, height, TextureFormat.RGBA32, false);
m_Texture = new Texture2D(width, height, TextureFormat.BGRA32, false);`

ckorris, 2019-12-04:
The very easiest way to deal with this is to swizzle the channels in the shader you use to display the texture, when you sample it. For example, see how we do it in ZED_Forward_Lighting, the main material used for displaying the ZED image in most scenes ([here](https://github.com/stereolabs/zed-unity/blob/6c0c719092facd9e95cb85af4941bd694c52b340/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/Lighting/ZED_Forward_Lighting.shader#L115)): 

`float4 color = tex2D(_MainTex, uv.xy).bgra;`

The swizzle is the ".bgra" we add at the end, and specifies which order to sample the channels from. 

If you don't have control over that shader, or don't want to modify it specifically for the ZED image, you can use [Graphics.Blit(Texture, Texture, Material)](https://docs.unity3d.com/ScriptReference/Graphics.Blit.html) to copy the texture into another texture, using a material that contains an image effect shader that does the exact same thing. 

It's also possible to convert the colors in a script on the CPU, but that's a very slow process, between having to grab data off the GPU, iterating on the entire image on a single or small number of cores, and then putting it back on the GPU. So I wouldn't recommend it.

dimll, 2019-12-04:
Thanks for the quick response @ckorris !

closed 2019-12-04 by dimll