stereolabs/zed-unity #98: Oculus Quest Link with ZED Mini

hankh1105, 2020-01-14:
I heard the Quest Link cable makes the HMD PC powered now, I'm curious if it is compatible with ZED unity development?

HyroVitalyProtago, 2020-02-17:
I'm currently trying to get raw images data in UVC mode with https://assetstore.unity.com/packages/tools/integration/usb-camera-for-unity-android-151744 but the framerate isn't stable for now, even in the lowest resolution... Don't know yet if it's a power consumption problem with OTG or the Frame Buffer technic that is not sufficiently efficient. Will post updates here.

DanielPD, 2020-06-04:
Can confirm that this works as expected in Unity 2019.3.15 with the latest Oculus integration and ZED 3.1.0: I'm using my Oculus Quest tethered to the PC with a USB cable, and A ZED Mini stuck to the front of the headset, also tethered to the PC with a USB cable. I'm getting all the passthrough AR goodness :)

(It also works well with VirtualDesktop streaming to the Quest. That way you only have 1 USB cable for the ZED going to the headset, makes it a bit more manageable)

KP021, 2020-06-06:
Thanks, @DanielPD Can you post more details about how you got it to work?

DanielPD, 2020-06-08:
Well, it actually just works as you would expect :D

First make sure you have the Quest updated and the Oculus software installed and updated on the PC (you might need to op-in for the beta release). Then set up the link connection between headset and PC, and confirm that it all works.

Then in your Unity project make sure to enable VR support: Project Settings -> Player -> XR Settings -> (Depricated) Virtual Reality Supported

(I haven't tried with the new XR Plugin Management system yet)

Import the ZED for Unity package, drag the ZED_Rig_Stereo prefab to the scene and press play. Unity should automatically recognise the Quest-through-link as a generic Oculus device and should output the stereo image to the headset.

Note that you don't need the Oculus Integration Unity plugins for the basic 3D camera images to work.

tommasorosi, 2020-06-08:
I tried it and it works, but the stereo images of the ZED felt just a bit weird, something I never felt with the Rift. This doesn't happen to me with virtual objects, only with the stereo images from the ZED. Anyone else feels like the experience is just a bit off?

KP021, 2020-06-08:
Thanks, @DanielPD for sharing. Did you try developing apps using both the ZED's video and VR environment?
@tommasorosi Well it may be because the eyes align with the ZED better on Rift while on Oculus the ZED may be vertically displaced from where your eyes would be.

cookiegobbler, 2020-06-10:
HMM so this can be potentially solved with a different mounting setup?

wsteelenyc, 2020-12-12:
Anyone done any experimenting with the quest 2 via link?

WHAMWONG, 2021-02-19:
I tried to do the experiment with the Oculus Quest 2 via link. Yes, it works. Once the quest 2 is connected to the PC via the link, the quest 2 switches to PCVR system mode, which actually gives very little difference from the Oculus Rift. The field of view of AR will be farther away when using the quest 2, compared to using Rift. I think it's because Stereolabs is not yet compatible with the quest 2. 

Using the new XR Plugin Management system will cause the zed camera to fail to open and still require Virtual Reality Supported to be turned on, as @DanielPD said.

keli95566, 2021-09-02:
Thank you guys so much for sharing this. I am using Unity 2019.4 LTS and Quest2. Quest2 connected to the PC succesfully via link or virtual desktop, however, when I run the scene from Unity editor, nothing shows up from Quest2. Does anyone have an idea what could be the mistake?

closed 2022-04-27 by github-actions[bot]