stereolabs/zed-unity #102: Unity examples（planetarium）：game show nothing

peter2527, 2020-02-04:
First the sdk and cuda is ok,and plugin is ok.
When i launch the example:Planetarimu!
When the play button not press  the the solar system is on the screen,but when i press the play button ,the solar system is covered by a white screen,meanwhile the audio belong to the scene is playing.
When i import the pluginpackage,there is warnning"Show Unity Splash screen(current=Ture)",is that the problem?

obraun-sl, 2020-02-04:
Hi,
The warning message is not a problem, it's just game configuration when building the app.
Are you in DX11 or OpenGL rendering ?

peter2527, 2020-02-04:
> Hi,
> The warning message is not a problem, it's just game configuration when building the app.
> Are you in DX11 or OpenGL rendering ?

I am a new guy,i just fellow the tutorial provided by the stereolabs,i know the warning is not the matter,but a can't apply the game,what's going on?

Bvallon-sl, 2020-02-04:
Hi,

obraun-sl, 2020-02-04:
For me, it looks like a problem in the direct GPU rendering (CUDA to DirectX or OpenGL). 
Can you make sure that : 
- your display is connected to the DVI/Hdmi/DP port of the NVIDIA graphic card (same than used by the SDK)
- your NVIDIA driver is up to date.

You can also test the ZED SDK C++ sample in C:/ProgramFiles(x86)/ZED SDK/Samples/bin such as depth sensing which is using CUDA - OpenGL interoperability. 
You can also use the Depth Viewer tools which also operates the same way.

peter2527, 2020-02-04:
ok ，i will have a try！appreciate

发自我的iPhone

peter2527, 2020-02-08:
> For me, it looks like a problem in the direct GPU rendering (CUDA to DirectX or OpenGL).
> Can you make sure that :
>
> * your display is connected to the DVI/Hdmi/DP port of the NVIDIA graphic card [...]

All the 11 samples in  C:/ProgramFiles(x86)/ZED SDK/Samples/bin worked very well！

peter2527, 2020-02-11:
> For me, it looks like a problem in the direct GPU rendering (CUDA to DirectX or OpenGL).
> Can you make sure that :
>
> * your display is connected to the DVI/Hdmi/DP port of the NVIDIA graphic card [...]

Hi！All the 11 samples in C:/ProgramFiles(x86)/ZED SDK/Samples/bin worked very well，but the unity have the same problem，for example，it looks like a white screen in front of the ZED_Rig_Mono.Please help!

closed 2022-04-27 by github-actions[bot]