stereolabs/zed-unity #103: From point cloud space to texture space (UV mapping)

dimll, 2020-02-11:
Hello, 

I have constructed a mesh, using zed’s point cloud values as given, cutting those who are above a specific depth value. I am trying to map each vertex of the mesh to the coordinates of the image texture. The texture that I am using is `colorTexture` from the ZEDPoingCloudManager.cs script.

When I get the point cloud values, I keep the point cloud coordinates(X,Y,Z) and the image coordinates (x,y) of the minimum and maximum point cloud elements that I am using to construct my mesh.
 
I tried to find the texture coordinates from the point cloud space to the image space using the below code. 
xyMinMaxPositions is a Vector3 array where:
xyMinMaxPositions[0] is the point cloud position of the element with the mimum X coordinate
xyMinMaxPositions[1] is the point cloud coordinate of the element with the mimum Y coordinate
xyMinMaxPositions[2] is the point cloud coordinate of the element with the maximum X coordinate
xyMinMaxPositions[3] is the point cloud coordinate of the element with the maximum Y coordinate

`
var pointCloudWidth = xMaxImageSpace - xMinImageSpace;
var pointCloudHeight = yMaxImageSpace - yMinImageSpace;
 for (var i = 0; i < 4; i++)
        {
            xyMinMaxPositions[i] /= xyMinMaxPositions[i].z;
        }
var projectedMeshWidth = xyMinMaxPositions[2].x - xyMinMaxPositions[0].x;
var projectedMeshHeight = xyMinMaxPositions[3].y - xyMinMaxPositions[1].y;

ax = pointCloudWidth / projectedMeshWidth / zed.ImageWidth;
bx = xMinImageSpace / zed.ImageWidth - ax * xyMinMaxPositions[0].x;

ay = pointCloudHeight / projectedMeshHeight / zed.ImageHeight;
by = yMinImageSpace / zed.ImageHeight - ay * xyMinMaxPositions[1].y;
`
and when assigning each UV in every vertex:

`
var a = new Vector2(ax, ay);
var b = new Vector2(bx, by);

for (var i = 0; i < vertices.Count; ++i) 
            {     
                var projectedCoords = new Vector2(vertices[i].x / vertices[i].z, vertices[i].y / vertices[i].z);
                var transformedCoords = a * projectedCoords + b;
                vertexUVs.Add(Vector2.one - transformedCoords);
            }
`
In the final part, I subtracted the transformed coordinates from one because the given texture was upside down.
 
Unfortunately, it seems to be an offset in the result and I can’t understand what I am missing. Below is an example of the mesh I am generating.
 
I created the mesh using the Marching Cubes algorithm based on the point cloud coordinates that I am getting from zed camera. 

[image]

Bvallon-sl, 2020-02-12:
Hi, 
you should project each triangle of your mesh to the image to warp it properly.
Using the camera parameters and the 3D positions of each vertices you can get the corresponding image point with the following formula:

sl.cameraParameters camera_parameter_scaled_to_image_size = zedcam.GetCalibrationParameters().leftCam  (camera parameters)

for each triangle:
 	for each vertice v:
		im_u = (v.x * camera_parameter_scaled_to_image_size.fx) / v.z + camera_parameter_scaled_to_image_size.cx;
		im_v = (v.y * camera_parameter_scaled_to_image_size.fy) / v.z + camera_parameter_scaled_to_image_size.cy;

dimll, 2020-02-13:
Thank you! @Bvallon-sl

closed 2020-02-13 by dimll