stereolabs/zed-unity #104: Black screen after updating to ZED SDK 3.0.2 for zed mini

kaszim, 2020-02-12:
After updating the ZED SDK and the unity plugins to the latest version none of the examples work. All binaries under "C:\Program Files (x86)\ZED SDK\samples\bin" works, as well as "C:\Program Files (x86)\ZED SDK\tools". There seems to be some issues with the updated unity plugin.

Everything worked fine under 2.8.

kaszim, 2020-02-12:
The issue seems to be within the ZED_Forward_Lighting shader. Simply overwriting the current version with https://github.com/stereolabs/zed-unity/blob/c1c206b2e50e1440f730b2c0765fd4aac63eaf83/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/Lighting/ZED_Forward_Lighting.shader resolves the issue.

obraun-sl, 2020-02-12:
Can you provide the Unity version you are using (2018.xx, or 2019.y) ?

kaszim, 2020-02-12:
I am using Unity version 2019.3 under Windows, with an Oculus Rift S (if that makes any difference for shaders).

kaszim, 2020-02-24:
Resolved by #106

closed 2020-02-24 by kaszim