stereolabs/zed-unity #107: Error in ChooseTrackedObjectMenu.cs

mattun, 2020-02-20:
When compile, appear error.

I checked 270 line of  ChooseTrackedObjectMenu.cs.

So I found CreateTrackedObjectPrefab.

```
CreateTrackedObjectPrefab(TOUCH_INDEX_LEFT, out ChooseTrackedObjectButton leftButton);
if (button != null)
{
    allButtons.Add(button);
}

CreateTrackedObjectPrefab(TOUCH_INDEX_LEFT, out ChooseTrackedObjectButton leftButton);
if (button != null)
{
    allButtons.Add(button);
}
```
then I edited like below

```
CreateTrackedObjectPrefab(TOUCH_INDEX_LEFT, out ChooseTrackedObjectButton leftButton);
if (leftButton != null)
{
    allButtons.Add(leftButton);
}

CreateTrackedObjectPrefab(TOUCH_INDEX_RIGHT, out ChooseTrackedObjectButton rightButton);
if (rightButton != null)
{
    allButtons.Add(rightButton);
}

```

So I could compliled. 
Is this right??

obraun-sl, 2020-02-21:
You are right. The merge with 3.0 fails with that part. 
I am pushing the fix now.

Thanks.

obraun-sl, 2020-02-21:
Note : The fix was already in master branch. Make sure to always pull on master branch to have the last up-to-date version.

closed 2020-02-21 by mattun

brandonjmatthews, 2020-03-04:
The Unity package that is published under the major release 3.0.0 appears to still have this issue. The release is for commit [3debac4](https://github.com/stereolabs/zed-unity/commit/3debac4ddf2886b55f3fef897c6fbc55b87a6a25) while the fix is on commit [f9ecc20](https://github.com/stereolabs/zed-unity/commit/f9ecc20eb3abe02ac9d2f0ce942da96f16efd97b).