stereolabs/zed-unity #109: ZED Unity Plugin v3.0.0's ZED_Rig_Stereo and ZED_Rig_Mono doesn't show anything in game scene

AkiraShibata18, 2020-02-25:
I tried to use ZED Unity plugin, that I put ZED_Rig_Mono in the new scene in Unity (2019.3.1f).
After playing the Scene, 'Loading Camera' was successful but nothing was shown in game scene (black screen).
I found when I changed the shader of frame object from ZED/ZED Forward Lighting (default) to ZED/Unlit RGBA, the image was shown properly. Thus I feel it's a problem of around the shader.
My versions are
- Device: ZED mini
- OS: Windows10
- ZED SDK version : v3.0.2
- cuda: 10.2
- ZED Unity Plugin: v3.0.0

Also, I tested in older versions below too, and it works well (show proper images in default settings).
- Device: ZED mini
- OS: Windows10
- ZED SDK version : v2.8.5
- cuda: 10.2
- ZED Unity Plugin: v2.8.2
Is there any changes around the shader or other pipelines?
Thank you.

obraun-sl, 2020-02-25:
Hi,

Please pull the master branch to the latest commit, since this has been fixed recently (https://github.com/stereolabs/zed-unity/commit/47a7026480064fa339c81553310d5b9fd4fbce67)

Best,
OB

AkiraShibata18, 2020-02-25:
It works!! Thank you very much!
I hope this will be applied to Release version soon.

closed 2020-02-25 by AkiraShibata18