stereolabs/zed-unity #110: No texture for spatial mapping

kaszim, 2020-02-25:
I am having some problems with spatial mapping under unity 2019.3. No texture is retrieved from the ZED as the call to the camera
```c
dllz_apply_texture(CameraID, nbVerticesInSubmeshes, nbTrianglesInSubmeshes, ref nbSubmeshes, updatedIndices, ref nbVertices, ref nbTriangles, textureSize, nbSubmeshMax);
```
returns textureSize with 0, 0. I have been unable to pinpoint what causes this. However the spatial mapping used to work under ZED SDK 2.8, and I couldn't see any large code changes to the spatial mapping for the unity plugin, so this might be an issue with the dll?

kralle1994, 2020-02-26:
I seem to have the same problem using Unity 2018.4. See the picture below. I do believe it is an issue with dll.

When saving the mesh a one pixel texture is made with the color (205, 205, 205, 205). Which if I am not mistaken is very close to the unity default material color. 

[image]

obraun-sl, 2020-02-26:
Hi,
Please check latest commit to see if this fixes the issue
https://github.com/stereolabs/zed-unity/commit/cc0a844da524896413b4228fc6efea9416996af8

kaszim, 2020-02-27:
@obraun-sl Sadly, the issue persists with the textureSize being returned as (0,0).

ckorris-imt, 2020-03-18:
Just FYI, `dllz_apply_texture` isn't meant to retrieve a texture - it's for applying the texture specified in the arguments to the scanned mesh. It only returns whether it's successful (I believe) and modifies the parameters set as `ref`, which do not include anything related to the texture. 

So on the breakpoint @kralle1994 set up, it is indeed a problem that the texture size is 0, 0, but that's not happening at that particular call; you're simply seeing what's being fed _into_ that method, not the result. 

I'm not equipped to test/debug this right now, but I would track this backward to see where that texture size came from (starting with the last few calls visible in the stack trace).

ckorris, 2020-03-18:
Just FYI, dllz_apply_texture isn't meant to retrieve a texture - it's for applying the texture that's being produced from the scanning process to the scanned mesh. It only returns whether it's successful (I believe that's what the bool is - not documented outside the .dll) and modifies the parameters set as ref, which does not include anything related to the texture.

So on the breakpoint @kralle1994 set up, it is indeed a problem that the texture size is 0, 0, but that's not happening at that particular call; you're simply seeing what's being fed into that method, not the result.

I'm not equipped to test/debug this right now, but I would track this backward to see where that texture size came from (starting with the last few calls visible in the stack trace).