stereolabs/zed-unity #112: Nothing is rendered after upgrading

thedevleon, 2020-02-26:
Just upgraded to the ZED SDK v3.0.3 with CUDA 10.2, and the latest release of the ZED Unity SDK v3.0.0.

However, now the cameras are not rendered at all anymore. All I get is a black screen in all my scenes, also including all the example scenes. I even set up a completely blank project, imported the latest Unity SDK, made a scene with just the ZED_Rig_Stereo, but all I get is a black image and a blank frame.

[image]

thedevleon, 2020-02-26:
After downgrading to SDK version 2.8.5b and the Unity plugin v2.8.3, everything works again.

obraun-sl, 2020-02-27:
Hi,
Make sure to use the latest commit on Master branch (https://github.com/stereolabs/zed-unity/commits/master)
This contains the fix for Unity 2019 black screen.

A new release will be tagged as soon as some fix are indeed validated, but the best is to always stay with the up to date master branch to have the latest fix.

closed 2020-03-08 by thedevleon