stereolabs/zed-unity #113: zedCamera.RetrieveImage causing crash

Abrahamh08, 2020-02-29:
I have two simple lines:

`sl.ZEDMat image_zed = new sl.ZEDMat();       ZEDManager.GetInstance(sl.ZED_CAMERA_ID.CAMERA_ID_01).zedCamera.RetrieveImage(image_zed, sl.VIEW.LEFT);`

When I run the project, I get an access violation from sl_zed.dll and Unity crashes. I am on Unity 2019.3.3f1

Abrahamh08, 2020-03-25:
The problem was that zed matrix default constructors give them a pointer value of 0 in the new versions of the plugin. I had to run the create method on my zed matrix before passing it into the RetrieveImage function.

closed 2020-03-25 by Abrahamh08

BAE-seongchan, 2020-11-15:
I really do need your solution on this problem..
What does the "run the create method" mean?? (I can't find where the definition of ZEDMat is.)
If you can attach your source code, please attach it...(because i am not so good at English..)