stereolabs/zed-unity #115: ZEDManager instantiated at runtime failes to create camera

thedevleon, 2020-03-01:
Long story short, I'm creating a ZEDManager at runtime to read from an SVO file, setting it up as follows:

```
        var virtualZedGO = new GameObject("VirtualZED");
        virtualZedGO.transform.SetParent(this.transform);
        virtualZedGO.SetActive(false);
        virtualZED = virtualZedGO.AddComponent<ZEDManager>();
        virtualZED.cameraID = ZED_CAMERA_ID.CAMERA_ID_02;
        virtualZED.inputType = INPUT_TYPE.INPUT_TYPE_SVO;
        virtualZED.enableTracking = false;
        virtualZED.enableSpatialMemory = false;
        virtualZED.svoInputFileName = currentRecording.SVOFileName;
        virtualZED.svoRealTimeMode = true;
        virtualZED.svoLoopBack = true;
        virtualZED.pauseSVOReading = false;
        virtualZedGO.SetActive(true);
```

However, this fails and gives me an error:
`ZEDManager on VirtualZedCamera couldn't connect to camera: CAMERA_ID_02. Check if another ZEDManager is already connected.`

I also have a ZED_Rig_Stereo with Camera_ID 1 in the scene that connects to my ZED Mini attached to my Valve Index just fine.

To further test why this happens, I also have a disabled game object in my scene that has the ZEDManager component added to it with the same parameters above but set in the editor.

If I run the scene with it deactivated, and I activate it a couple of seconds after the scene loads, I get the same error.

However, If I enter play mode with the game object and the attached ZEDManager already enabled, it works just fine, I get no error, and I can see the SVO playing and the currentFrame counter is looping as it should. 

Maybe this is some kind of race condition?

The reason for creating the ZEDManager at runtime in the first place is because I want to play back different SVO files at runtime selected by the user, and to then display the PointCloud of these recordings using the ZEDPointCloudManager. Seems like there is no support for hotswapping the current SVO, as the SVO file is passed in the initialization of the ZEDCamera.

thedevleon, 2020-03-02:
Another issue I found:

When real-time mode is enabled on the second ZEDManager, it starts lagging immensively as soon as the first ZEDManager (ZED_Rig_Stereo) is fully initialized, and even though looping is enabled, the ZEDManager stops playback at some point.

Here's an video that shows the issue.
Notice that while, and before the ZED_Rig_Stereo has fully loaded, playback of the second ZEDManager (visualized with a PointCloud) is smooth. As soon as the first ZEDManager (ZED_Rig_Stereo) has initialized, playback starts lagging, and ultimatively, stops.

https://youtu.be/QMEa9shio_Q

zawazawagh, 2021-02-07:
Having the same problem with  SDK version 3.4 & Unity plugin version 3.4

closed 2022-04-27 by github-actions[bot]