stereolabs/zed-unity #116: Virtual objects not rendered in AR with OpenGLCore

thedevleon, 2020-03-02:
Hi,

I'm currently modifying the included ZED PointCloud Shader to be able to change the Size of the rendered points.  For this I'm using psize, which is only available in OpenGL, but not DX11. It works, but there is one big issue.

As soon as I switch over to OpenGLCore, no virtual objects are rendered in the ZED_Rig_Stereo anymore. Here's the exact same scene, once with DX11, once with OpenGLCore.

DX11:
[image]

OpenGLCore:
[image]

I'm using SDK version 2.8.5b and the Unity plugin v2.8.3.

ckorris, 2020-03-16:
I used to work on the plugin and haven't touched it for a bit, but, my feeling is that you need to invert how the depth works in the shader. 

Note [this code](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/ZED_Utils.cginc#L43-L47) in the ZED_Utils.cginc file: 

```
#if SHADER_API_D3D11
	colorXYZ.g = -colorXYZ.g;
#elif SHADER_API_GLCORE
	colorXYZ.g = -colorXYZ.g * 2 + 1;
#endif
```

That's from a function used to get the xyz position from a pixel. If I recall, it's only used by the point cloud shader. But [here's the Z-only version](https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Shaders/ZED_Utils.cginc#L77-L82) that's used in the main shaders used for typical rendering: 

```
#if SHADER_API_D3D11
	colorXYZ = -colorXYZ;
#elif SHADER_API_GLCORE
	//colorXYZ = -colorXYZ * 2 + 1;
	colorXYZ = -colorXYZ * 2;
#endif
```

Note the line with the +1, like it appears in the 3D version, was commented out. I thiiiink I did that a long time ago. Could be worth making the same change to the 3D version. 

Again, I'm out of practice, and also just trying to throw in my two cents before work, but I can try taking a closer look at some point if no answer comes up.

ckorris, 2020-03-18:
I tried replicating this but couldn't. I'm using SDK 3.1, but looked at the ZED_Forward_Lighting shader and supporting ZED_Utils file and nothing caught my eye that could make a difference here. 

Does this happen with a fresh new project? With nothing but the ZED plugin imported, set to OpenGLCore, and a scene with nothing but a ZED prefab and cube? I'm trying to be sure I'm looking where I should be.

thedevleon, 2020-04-02:
> I tried replicating this but couldn't. I'm using SDK 3.1, but looked at the ZED_Forward_Lighting shader and supporting ZED_Utils file and nothing caught my eye that could make a difference here. [...]

I'm sorry for the late reply, I only just now got around to updating to the newest ZED SDK and Unity plugin. The issue still persists. ZED SDK 3.1.2 and Unity plugin 3.1.0.

I have attached a simple project that just has the newest ZED Unity SDK imported, a Sample Scene with the ZED_Rig_Stereo, a simple cube, and the Graphics API set to OpenGLCore. If you run it, you'll only see the view from the ZED, but not the cube. As soon as you switch back to DirectX, restart and run the scene again, the cube is shown.

https://drive.google.com/file/d/1DQQzGXSo7nfEuZ-yg4ABFGCZ_ULMEkV9/view?usp=sharing

closed 2022-04-27 by github-actions[bot]